wled/WLED #249: mqtt state in json-format?

lebherz, 2019-10-09:
Hi!

its possible to respond the [mqttDeviceTopic]/v not in XML but in json format?
Its very easier to implement all the states in Home-Assistant.

best regards und beste Grüße
René

Aircoookie, 2019-10-10:
Hi René!
Yes, sure, this is being worked on! See #207.
Since I believe the full rework will still take a bit, I believe I will implement the [mqttDeviceTopic]/api and [mqttDeviceTopic]/api/set topics beforehand.

Viele Grüße :)

lebherz, 2019-10-10:
maybe a the output of of the Json-API
https://github.com/Aircoookie/WLED/wiki/JSON-API
in  [mqttDeviceTopic]/v/json would be nice :)
Grüße

anarchking, 2019-10-18:
Would it be possible to use serializeSegment serializeState serializeInfo to Mqtt publish states to topics Seg, Stat, Info? Trying to find a easy fix until the reworks :)

lebherz, 2019-10-19:
Hi @anarchking 
definitely yes! 
It's a lot of Data but all what we need in Home Assistant :)

regards René

lebherz, 2019-10-19:
... and a possible fast solution :) 👍 🥇

Matten-Matten, 2021-09-23:
Hello, is there any progress in implementing the json API in MQTT?

I already use WLED successfully via MQTT in Redmatic (nodered on HomeMatic).
Unfortunately I can't control everything via MQTT yet and have to do it via an HTTP call.

greeting
Matten Matten

blazoncek, 2023-01-25:
You can control every aspect of WLED through JSON formatted messages.
Just send them to `wled/device/api` topic.

closed 2023-01-25 by blazoncek