wled/WLED #251: Random colours on SK6812

lostdave, 2019-10-11:
More of a place holder here, and incase someone else has seen this.
I am getting random colours on Pixels when running sk6812.  
Built from source with atom
I will need to get the scope out at somepoint to see if there is anything there that shouldn't be.

Also getting alot of device unreachable from the app on this build

Aircoookie, 2019-10-12:
Are the random colors persistent or are there more like random flashes once in a while? In that case, this would be #230 .

Thanks for the feedback with device unreachable. That seems to pop up now and again even though the device is in face online, need to see about that.

lostdave, 2019-10-14:
One or 2 pixels will display an incorect static color, it almost feels like those pixels are incorrectly ordered in their rgb, but works fine with fastled tests.

I will do some more testing and let you know.

lostdave, 2019-10-15:
OK...Further looking at this.
I have one red Pixel that will walk it's way down the strip.
And then I have 3 pixels that will start on a random colour and stay there.
Even with brighness turned down...
The latter sounds like a strip problem.

sansillusion, 2019-10-15:
Do you have clock overlay activated in options ?

lostdave, 2019-10-15:
All Good,
Aircookie worked out.
Not a bug with wled.
The Wemos I was using had previously had Tasmota on it.
I had OTA flashed back to minimal and then wled over the top.
There must have been a hang over from tasmota in there doing something.
As soon as I did flash_erase wtih esptool, and reloaded wled, all was good in the world.

closed 2019-10-15 by lostdave