wled/WLED #253: Enhancement for HomeKit

Partizan7676, 2019-10-12:
It is good idea if your project will support HomeKit

Aircoookie, 2019-10-13:
Hi, I don't believe that HomeKit offers any local API right now to achieve this. Would be cool though, so I'll keep this issue open!

creativeretail, 2019-10-14:
Here you go:-

https://developer.apple.com/support/homekit-accessory-protocol/

It's free for non-commercial use. The specification is only 259 pages long ;)

Partizan7676, 2019-10-15:
> Hi, I don't believe that HomeKit offers any local API right now to achieve this. Would be cool though, so I'll keep this issue open!

Thanks! But may be that example will help you https://github.com/maximkulkin/esp-homekit-demo/tree/master/examples/led_strip_animation . I ask author to add AP-mode for settings but no result. This example gives you two devices - one of them is for control colour and brightness, other is for choosing effect by changing colour and animation speed by changing brightness.

dbartelmus, 2019-11-21:
Bump to make it happen. There is a new thing around ESP and HomeKit with OTA updates called HAA - the author from raven core is responsible for it.
HomeKit will be like awesome.

ch0w7, 2019-12-16:
would love to integrate with https://github.com/RavenSystem/haa

TheOriginalMrWolf, 2020-10-25:
*BUMP!!*

Might be worth having a look at this HomeKit implementation using Arduino core & WolfSSL: https://github.com/Mixiaoxiao/Arduino-HomeKit-ESP8266

Efficient, reasonably stable, reasonably performant (after pairing which takes about 30s due to all the crypto), low heap use - but need to set CPU to 160Mhz.

Aircoookie, 2020-10-25:
That project indeed seems promising! Seems like we'll be able to integrate it as it was designed to save enough heap to run other things.

The best way to integrate it would be first as an usermod (should just be a matter of modifying [this example](https://github.com/Mixiaoxiao/Arduino-HomeKit-ESP8266/blob/master/examples/Example05_WS2812_Neopixel/Example05_WS2812_Neopixel.ino)) and then, if it works well, adding it to the WLED code base both with a #define (to save the program and heap size if you don't use it) as well as a runtime setting (to remove the crypto processing times if you have it compiled in but don't use it).
There is luckily also an ESP32 version of the library with the same API and of course better performance.

I will see if I can get around to it myself at some point but if someone wants to do a PR with a usermod I'd be very grateful 👍

Aircoookie, 2020-10-25:
Just found an issue though and that is that the library utilizes the first 1408 bytes of EEPROM to save the keys, which is already occupied by the WLED settings and presets. We will only be able to integrate it after we fully moved settings and presets to the filesystem in the upcoming v.0.11.0

TheOriginalMrWolf, 2020-10-25:
Bring on v.0.11.0 !! 👍 :)

compgeek35610, 2020-12-25:
Now that V.0.11 is out, this would be a great feature to see. I know there are plenty of LED strip firmwares for ESP8266 that already support HomeKit but they leave a lot to be desired.

julieplz, 2020-12-27:
Looks like expressif released native support for HomeKit for ESP32, might come in handy?

https://github.com/espressif/esp-apple-homekit-adk

ErrorErrorError, 2022-03-04:
Hello! 

I am currently implementing support for native HomeKit support w/ WLED with the help of a stripped down version of https://github.com/HomeSpan/HomeSpan in order to save space. 

However, this will be only supported for ESP32 due to having more processing power to calculate encryption and decryption compared to an ESP8266.

ErrorErrorError, 2022-06-23:
Update: 
I have implemented HomeKit on a fork of WLED on my repo a while back but haven't gotten the chance to improve space, also haven't checked how much space it will take. 

However I did follow how y'all implemented `espalexa` so it does not include the homekit portion if you don't want to include it using `#define`.

maiiikell1995, 2022-06-24:
@ErrorErrorError this would still only be an ESP32 implementation correct?

ErrorErrorError, 2022-06-24:
> @ErrorErrorError this would still only be an ESP32 implementation correct? 

Yes, due to the ESP8266 does not have a dedicated hardware to calculate encryption/decrption fast enough while running WLED.

shauder, 2022-09-28:
@ErrorErrorError I tried compiling your fork and flashing it but I cannot get the ESP32 to boot and broadcast the AP after flashing. I am setting it to the esp32dev target. Any suggestions?

ErrorErrorError, 2022-09-28:
> @ErrorErrorError I tried compiling your fork and flashing it but I cannot get the ESP32 to boot and broadcast the AP after flashing. I am setting it to the esp32dev target. Any suggestions?

Hm, is there any logs? Can you see the serial monitor and see what's outputting?

shauder, 2022-09-28:
Nothing very useful unfortunately 

```
0:38:26.484 -> % ⸮ʀ#l`~⸮l⸮lrl⸮oo⸮⸮rob#⸮⸮l⸮⸮⸮⸮⸮`⸮o{⸮⸮b⸮bc⸮⸮⸮b⸮cc⸮⸮|⸮bcp⸮⸮⸮|⸮cbn⸮⸮|⸮cb⸮⸮⸮|⸮bl`r⸮o⸮⸮⸮⸮⸮~⸮c⸮oopn⸮⸮bll`p⸮o⸮b⸮⸮r⸮l⸮p⸮l⸮lll⸮⸮⸮l⸮⸮⸮lrl⸮s⸮bc⸮l⸮⸮⸮⸮lllr⸮⸮⸮lp⸮⸮oo⸮⸮⸮bl⸮l⸮⸮⸮⸮⸮⸮l⸮⸮l`~⸮l⸮lrl⸮on⸮⸮robc⸮⸮l⸮⸮⸮⸮⸮`⸮or⸮⸮b⸮bc⸮⸮⸮b⸮cb⸮⸮|⸮bbp⸮⸮⸮|⸮cco⸮⸮|⸮bc⸮⸮⸮|⸮bl`r⸮n⸮⸮⸮⸮⸮~⸮c⸮onpn⸮⸮bll`p⸮o⸮br⸮⸮⸮⸮lco⸮|b⸮lp⸮o⸮br⸮⸮⸮⸮l⸮bo⸮|bl⸮⸮lp⸮n⸮b⸮⸮bo⸮|bl⸮⸮lp⸮o⸮b⸮⸮bn⸮|cl⸮l⸮⸮l⸮n⸮|⸮⸮⸮⸮⸮⸮`Ada
```
Then I hit reset button
```
20:38:32.749 -> ets Jul 29 2019 12:21:46
20:38:32.749 -> 
20:38:32.749 -> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
20:38:32.749 -> configsip: 0, SPIWP:0xee
20:38:32.749 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
20:38:32.749 -> mode:DIO, clock div:1
20:38:32.749 -> load:0x3fff0018,len:4
20:38:32.749 -> load:0x3fff001c,len:1044
20:38:32.749 -> load:0x40078000,len:8896
20:38:32.749 -> load:0x40080400,len:5816
20:38:32.749 -> entry 0x400806ac
20:38:33.231 -> Ada
```

I have 115200 baud set

ErrorErrorError, 2022-09-28:
> Nothing very useful unfortunately
>
>
>
> ```
>
> 0:38:26.484 -> % ⸮ʀ#l`~⸮l⸮lrl⸮oo⸮⸮rob
>
>
> #⸮⸮l⸮⸮
> ⸮⸮⸮`⸮o{⸮⸮b⸮b
>
> 
>
> c⸮⸮⸮b⸮c
>
> 
>
> c⸮⸮|⸮b
>
> 
>
> cp⸮⸮⸮|⸮c
>
> 
>
> bn⸮⸮|⸮c
>
> 
>
> b⸮⸮⸮|⸮b
>
> 
> l`r⸮o⸮⸮⸮ [...]

Make sure you clear everything in flash. If that doesn't work then we'll, I haven't touched this project in a while, but it was working before I stopped working on it

shauder, 2022-09-28:
Looks like I got it to boot by flashing the boot loader again, weird! I don't see the HomeKit settings so I am guessing I need to recompile with a flag?

ErrorErrorError, 2022-09-28:
> Looks like I got it to boot by flashing the boot loader again, weird! I don't see the HomeKit settings so I am guessing I need to recompile with a flag?

It's not a setting, you'll need to connect it to The wifi, and once you do I'll be available to add on homekit. The passcode is hard coded so it should be somewhere ;p

shauder, 2022-09-28:
Thanks for the replies! I have not got it to pair successfully yet but the code is "87766968" for anyone else who comes across this.

ErrorErrorError, 2022-09-28:
> Thanks for the replies! I have not got it to pair successfully yet but the code is "87766968" for anyone else who comes across this.

What does the logs show when it tries to pair?

shauder, 2022-09-28:
This is all I see in the serial monitor when I try and add it. I can see it in the list, then I can enter the code and it seems to like it then it errors on "Accessory is not reachable" but I am able to refresh the WLED UI the entire time it's trying.

```
21:47:41.700 -> E (35688) wifi:AP has neither DSSS parameter nor HT Information, drop it
21:47:45.372 -> E (39391) wifi:AP has neither DSSS parameter nor HT Information, drop it
21:47:48.787 -> dhcps: send_nak>>udp_sendto result 0
```

I am not sure if there are better logs somewhere that I can look at.

blazoncek, 2022-09-28:
@ErrorErrorError please link to an issue on your fork and continue discussion there until the code is ready for PR.

When the code is ready use [these instructions](https://github.com/Aircoookie/WLED/wiki/How-to-properly-submit-a-PR) to submit PR.

defektu, 2024-01-11:
Hi all, I managed to connect homekit with another spare ESP8266 to WLED without homebridge. It just reroutes homekit request to WLED via WebSocket. With full hsv and brightness control ofc
You can set multiple homekit devices with single ESP8266 with routing. Works like cheap homebridge.

mkormendy, 2024-06-24:
@defektu care to elaborate on the steps?