wled/WLED #254: Hue Sync not working

XXXAAAMMM, 2019-10-12:
Hello,

I am currently implementing wled into my home lighting. Until now everything works flawlessly. This really is an amazing project. 
But since I own a few Phillips Hue lights I would love to sync my Led Stripe to them.
Unfortunately everytime I try to connect my Hue Bridge the Wled Software responds with a Json Parsing Error.

Is there a way to solve this problem?

Greetings from Germany
Max

Aircoookie, 2019-10-13:
Hi Max,

have you set the correct hue bridge IP address and Light ID number of the Hue bulb you would like to sync your stripe to? You can find it in the Details section of the hue app. Also make sure you press the bridge button before saving the settings page. But I believe you did all that correctly, because you would get a different error code otherwise.

JSON parsing error means that your hue bridge most likely sent something the ESP didn't expect. I've just tried re-pairing with the latest version of WLED and hue bridge software, works fine.
You could help me find out what the issue is by enabling `#define WLED_DEBUG` in wled00.ino.

Viele Grüße auch von mir :)
Christian

XXXAAAMMM, 2019-10-28:
Hi Christian,

sorry for the late response. The Notification Mail from GitHub somehow landed in the spam folder.
I got it working now. I wasn't running the newest version on my Esp8266. After updating, everything worked flawlessly on the first try.

Thanks for your awesome work. This is an amazing project.

Beste Grüße
Max

closed 2019-10-28 by XXXAAAMMM