wled/WLED #255: ESP8266 seems to hang for 2-7 secs

fribse, 2019-10-12:
Hi Aircookie
Great job with this, I've connected a string of 210 LED's (SK6812).
I have a 5V 20A PSU connected to both ends of the string.

I have connected a USB PSU to the ESP12E.
One of the GND is connected to the PSU's lowvoltage gnd.
The D4 is connected to the LED data pin.
The D6 is connected to the PSU relay, to eliminate standby power.

When I run effects they sometimes freeze for a few seconds, and then continue, the same goes for colorchange, it takes some seconds once in a while.

It does react, but sometimes it takes a while for it to do so, what is wrong?

Aircoookie, 2019-10-13:
Hi!
do you have enabled Blynk, MQTT or Hue sync? Those services could cause such freezing.
If this happens, does the entire effect freeze or does it just take a while before the effect/color changes?
Thanks for your help!

fribse, 2019-10-13:
Hi Aircookie
Thankyou for getting back to me on this!
I only have MQTT enabled, as it is integrated to HA.
The entire effect freezes, it also has some delays in switching sometimes.

GsynthChris, 2019-10-14:
Hey Fribse, have you tried connecting only a strand or two and testing it as you add one more at a time?

fribse, 2019-10-14:
Sadly, I can't do that, as it is mounted in alurails with a diffuser glued on top of it and with shrinktubing around the soldered connectors, and then mounted under the bed, so it would be quite complicated
It's a 4 m strand, with 209 led's (were 210, but I accidently pulled the copper from the last led with the power injection when mounting them under the bed).
You can see it mounted and running here: https://youtu.be/7zTod2vpCwM, all though it's a very short clip, and with no freezing in it.

fribse, 2019-10-15:
I tried removing the mqtt config, but that didn't improve it.
Can it be a problem with the 3.3v version 5 v required by the SK6812 when it's 210 leds?
Or can the datachanel be flaky, and that will freeze the led chain?
I have no idea why, it's a bit sad, a LOT of time went into this project (way too much :-D).
I've just gotten my quindor diguno boards, and I will add such a board to this when I get the components, maybe that will help?

Marcmk1, 2019-11-16:
Hi,
I got the exact same issue on an d1 mini pro.
Flashed with 0.8.6
Created an setup with 60
2812b leds

Marcmk1, 2019-11-16:
> Hi,
> I got the exact same issue on an d1 mini pro.
> Flashed with 0.8.6
> Created an setup with 60
> 2812b leds

Issue solved when turning off
"Get time from NTP server"

fribse, 2019-11-17:
> https://github.com/Aircoookie/WLED/releases/tag/v0.8.6

That IS interesting, I have to check that when I get home.

Marcmk1, 2019-11-17:
> > Issue solved when turning off
> > "Get time from NTP server"
> 
> That IS interesting, I have to check that when I get home.

I have to say the my esp is blocked from internet connection by my firewall/router.
And that I was looking to setup my own ntp server.
I still need te dive in to this option.

fribse, 2019-11-17:
I have a local NTP server, and mine is also blocked from internetaccess.
But removing the ntp settings seems to have solved the problem. I've tested for 20 mins, and I will test further, but that would be a great solution.
The finding also corresponds with the other test setup I have, which is my first quindor diguno board, because I haven't set NTP on that, and it doesn't freeze at all.

fribse, 2019-11-17:
I tried enabling it on both my esp32 and my esp8266, and the esp8266 (the one that freezes) doesn't get the time updated properly, whereas the esp32 does get the time, and doesn't freeze up even with the ntp activated.

closed 2020-02-22 by fribse