wled/WLED #257: Set preset for part of the strip

oxodao, 2019-10-13:
Hi,

I have a 150 leds strip "splitted" in three part. Is it possible to create "groups of leds" to set a preset to ?
Like if I currently want to use only use from 0 to 120 with a preset and keep the last ones off, or if I want to light just behind my couch by lightning leds 50 to 70 ? Is there a way to do this with WLED ? If not, do you plan on adding it ? That would be a really useful feature

Anyway this is a really great piece of software, good work!

Aircoookie, 2019-10-13:
Hi!

Yes, this feature is called "segments" and I have been planning to add it for a very long time! I hope it will be available before the start of the holiday season!

oxodao, 2019-10-13:
Nice 👍 
Can't wait to play around with it, good luck to do this.

closed 2019-10-13 by oxodao

toto79, 2019-10-15:
today basic segements (up to 10) with individual animations are already implemented and functional for me... see wiki for more details: https://github.com/Aircoookie/WLED/wiki/JSON-API
("only" web config of this segments isn'n available yet)

oriolism, 2019-10-19:
This would be great! Im planning to install leds around the house for Christmas and the segments will simplify my setup. Thanks in advance!