wled/WLED #258: No wifi after data pin connected

sladaar, 2019-10-13:
Hi! First of all many thanks for the project!

I tried it as soon as i found, but there's this little strange thing: the nodemcu works well without D4 pin connected, via web browser or app. But then when i plug in the two led strip pins (data to D4 and ground) the nodemcu can't get wifi no more. 

Any suggestions?
Thanks in advance!

Aircoookie, 2019-10-13:
Welcome!
Just to clarify, you have connected D4 to data in and ground to ground, right? When you turn on the ESP with the data wire connected, do the LEDs turn on to their orange boot color? How long is your data cable? Please also try another NodeMCU if you have one, it might be a hardware fault.

sladaar, 2019-10-13:
Thanks for the reply Air!
Yes, D4 to data and groud to ground. Just for the info, i have other similar installations in my house using other sketches (mclighting, hyperion, custom based), but i really liked yours so i wanted to try it!
So, leds won't turn on at all on boot, data cable is about 20cm and i tried two new nodeMCUs (branded Amica, both).
To recap: if i only power on the board without connections, it works normally; if i power it on with the two duponts (data and ground) i can't reach it no more (boot failed?)

Do you think it may be due to interference on D4? I tried to avoid it, i guess it's more other stuff.
Thanks again!

raoulteeuwen, 2019-10-13:
What is the config of the LED-strip (nr of leds etc) and your power supply?

sladaar, 2019-10-13:
thanks raoul!
Led is a short ws2812b, like 30leds. Power supply 5v60A....i used it in the hyperion ambilight project, then unmounted all the rig!

raoulteeuwen, 2019-10-13:
Hi @sladaar . You seem to be more experienced than i am, so hope i won't be saying to many n00b things... I googled the NodeMCU Amica and found [this](https://roboindia.com/tutorials/nodemcu-amica-esp8266-board-installation/). It says it uses a CP2102 instead of a NodeMCU with a ch340g. Could that be playing a role? Also, could you confirm what pin on the Amika (if the one i linked seems the one you are using) your ground is connected to? Any change you also have a Wemos D1 mini laying around? I myself have WLED running on both a NodeMCU (with ch430g) as well as Wemos, fyi.

sladaar, 2019-10-13:
Hey @raoulteeuwen , many many thanks for the help, i didn't think about the controller! Didn't have problems with those two Amica boards before but your hint may be useful!
And to try with WemosD1 would be the work of tomorrow, already in schedule, it's 9pm here and i'm too tired to solder! 
So i'll post my results tomorrow in the afternoon, hoping to be helpful for someone.

Have a good evening

sladaar, 2019-10-14:
Ok for posterity: whole led strip faulty, not only the first chip.
And now it works on both WemosD1 and Nodemcu (branded Amica).
So simply, my bad of not checking the full strip.

Thanks again!

closed 2019-10-14 by sladaar

script-anzcro, 2019-11-30:
I observed a similar problem.
External LED power supply 12v20A 
ac to 5v to power the d1mini.

Example of my situation
Not Working:
If I connect the data wire to my led Strip with the power supply off , and common ground from d1 mini and the ground of the 12v power supply d1 mini doesn´t connect to the wifi .

Working
If I boot with the data wire connected to Led strip and without the common ground between d1mini and 12v power supply and connect the common ground later (after boot) everything works

If I boot without the data wire connected and with the common ground between d1mini and 12v power supply, and connect it later (after boot) everything works fine.

I use the pin D6 to trigger the start of my power supply (with a solid state relay). If you have any idea how to fix this behavior it will be great.

Thanks

totomaze, 2019-11-30:
I observed a similar problem.
External LED power supply 12v20A
ac to 5v to power the d1mini.

Example of my situation
Not Working:
If I connect the data wire to my led Strip with the power supply off , and common ground from d1 mini and the ground of the 12v power supply d1 mini doesn´t connect to the wifi .

Working
If I boot with the data wire connected to Led strip and without the common ground between d1mini and 12v power supply and connect the common ground later (after boot) everything works

If I boot without the data wire connected and with the common ground between d1mini and 12v power supply, and connect it later (after boot) everything works fine.

I use the pin D6 to trigger the start of my power supply (with a solid state relay). If you have any idea how to fix this behavior it will be great.

Thanks

Aircoookie, 2019-11-30:
@totomaze one other thing you could try that will most likely work is using a pullup resitor (anything around 4.7kOhm will work fine). Connect that between D4 and the 3.3v pin. Most likely the LEDs are pulling D4 low slightly, and it needs to be high for the ESP to boot :)

raoulteeuwen, 2019-11-30:
> and common ground from d1 mini and the ground of the 12v power supply d1 mini  .

It would be nice to exactly know how you connected the grounds (often, you, as the poster with the connection in sight, think the description is clear, but i do wonder whether the ground is set up correctly). Like: is there a ground wire running from your 12V power to the ground on the LED strip, and a wire running from the ground on your d1 to the ground of the LED strip, so the grounds 'meet' at the ground pin of the LED strip? Or is it wired differently?

totomaze, 2019-12-01:
The 4.7k resistance did the trick. Thanks for that.

ilandllc, 2020-09-22:
@totomaze  - super n00b question here but I'm having the same issue.  Did you just solder the 4.7k resistor between the D4 and the 3.3v pin?

totomaze, 2020-09-22:
Yes