wled/WLED #259: Binaries WLED_0.8.5_ESP8266_APA102.bin with BGR support :-)

DuschdrBabbe, 2019-10-14:
Hello,

need bin file for ESP8266 with the LED (APA102) setttings for BGR ...

THX

Pascal

lostdave, 2019-10-15:
not really an issue, but here you go.
Maybe discord would be a better place to request builds...or you could learn to do it yourself.
There are plenty of tutorials around for atom/arduinoide.

https://drive.google.com/open?id=1LUZZov9XfHM0LC9F6WNKOtfRAyUIu3Ve

DuschdrBabbe, 2019-10-16:
Hello Dave,

thank you so much ... yes discort is a option ... ;-) 

Pascal

DuschdrBabbe, 2019-10-16:
Hey Dave,
there ist no BGR in the list :-)

Greetings

Aircoookie, 2019-10-16:
Hi, i believe the binary Dave sent you has BGR coded in, so if you select GRB you'll have the correct colors. Regardless, i'll add BGR to the options list :)

DuschdrBabbe, 2019-10-16:
no sir .. ;-) have try it but red and blue are still reversed...
But THANKS for implementing!!!!

DuschdrBabbe, 2019-10-16:
sorry guys,
it,s my fault... wrong hyperion config...
it works...

closed 2019-10-16 by DuschdrBabbe