wled/WLED #261: Add delay in connecting wifi

sujitrp, 2019-10-15:
Please guide me to add delay in wifi connection. My router takes 5-6 min for boot and start normal.
Router is provided by the provider so no option to change the same.
In case off Power, outage scenario  WLED fails to connect Router and secure IP . 

Is any way can delay the WLED wifi connect function?

Aircoookie, 2019-10-17:
Hi! I am currently in the process of reworking the wifi connection logic. The new code will continuously try to connect and not stop trying after half a minute like the current version. So that will fix this problem :)

sujitrp, 2019-10-18:
Awsome Aircoookie

brentbrooks70, 2019-10-23:
@Aircoookie Have you considered using [Arduino_ConnectionHandler](https://github.com/arduino-libraries/Arduino_ConnectionHandler)? It does all the heavy lifting, like, loading appropriate drivers per board, connecting, notifying of successful connection via calling function, notifying (also via function call) that there has been a disconnect and even reconnects when available....in the connect/disconnect functions you can stop polling OTA and MQQT while not connected and resume them again once connected. This works on all Arduino SamD boards as well as ESP8266 and ESP32. I believe I had to install Arduino debug library as well but if it fits with your memory requirements it might be better than rewriting it yourself or at least look into their code and methodology of handling connections. I am using this on several boards that are connected to Arduino's beta cloud program (mostly MKR1000s, MKR1010s,  MKR1400s & 1 ESP8266) but do have the ConnectionHandler running on numerous esp boards and this allows code to be more portable across architectures. Just a thought especially if you start storing/running HTML & config from SPIFFS your code would work on any of the Arduino platforms as well ESPs with a few #ifndef statements.

Again, awesome project, can't wait for seg support to be fully implemented, so this is a completely selfish recommendation as I would rather you work on it than WiFi :) Have a good one!

Physikus74, 2019-11-24:
Hello Aircoookie,
I would be very pleased about the function that the software permantent tries to log on to WiFi. When will it be available? 
WLED "lights up" our living room for some time now and we are absolutely happy!

Aircoookie, 2019-11-24:
@Physikus74 have you already updated to 0.8.6 or were you experiencing issues with it? 0.8.6 has the new connection logic, so it will permanently attempt to reconnect!

Physikus74, 2019-11-25:
no, I still have 0.8.5. Thanks for the hint, I will update and report in the next time. ;o)

sujitrp, 2019-11-26:
With 0.8.6 update this issue is resolved

closed 2019-11-26 by sujitrp