wled/WLED #262: issue with iPhone app

qubie2012, 2019-10-15:
Hi I know this is not an issue with WLED code but i wasnt sure where to go with this one
The issue i have is the 16 preset buttons 1 - 8 work fine but 9 - 16 are underlined and if i try to use 9 - 12 my phone wants to call 910 1112 if i try 13 - 16 it wants to call 131 415 16.

I have tried a ANDROID phone and the presets are all fine just IOS 

by the way thankyou for this project I am have more fun playing around with this than i have in a long time.

tube0013, 2019-10-15:
I can confirm this happens for me as well. (Just upgraded to 13.1.3).

Hitting the WLED ip via safari does not have the same issue.

App:
[image]

Safari:
[image]

lostdave, 2019-10-15:
Same Same - 
I don't use the presets, so not so much an issue for me.

Aircoookie, 2019-10-17:
Interesting issue! That safari even considers this a phone number even though the digits are separated by HTML tags... Will be fixed by upcoming UI update!

qubie2012, 2019-10-19:
Thanks Aircoookie this is an awesome piece of work
love it.

doug1376, 2019-10-30:
Also have this issue on iPad and iPhone.  To add to it.  I also have no scroll bar on the right had side of FX and Palettes in iOS.  However Love this app.

Thanks aircookie
[image]

fijit586, 2019-12-04:
I have seen this too. iPhone8.

Ve6hgw, 2019-12-21:
I have this issue too but it’s just a very minor inconvenience.

closed 2020-04-26 by stale[bot]