wled/WLED #263: Fire TV Cube 2019 Alexa could not find device

kaiomatico, 2019-10-16:
Hi, first of all thank you for the awesome work! Sorry if it was already reported that the Fire TV Cube does not work because it is no "real" echo device, but I could not find any closed issues regarding this. Anyway I have the new 2. generation 2019 Fire TV Cube. I flashed the binary release of WLED 0.8.5 to my esp8266, configured the wifi, rebooted and everything is working fine. However I am not able to find my esp8266 as emulated hue device. When I ask the Fire TV ( which has an Amazon Echo built in) to search for new devices on my network it is not able to find it.

Is this a known problem? Can I help debugging with some logs or output? Thank you in advance!

kaiomatico, 2019-10-16:
Just a quick follow-up: The home-assistant emulated hue module also seems to not work.

Aircoookie, 2019-10-17:
Thanks for the report! Seems like the Fire TV behaves differently to other Echo devices...
That the home assistant module doesn't work either makes me believe it is an issue with the Echo not sending some required packets. You could try defining `#define ESPALEXA_DEBUG` at the top of wled00.ino and watching the Arduino serial monitor log while trying to discover devices. I've precompiled a test binary for you that will send the debug log to the serial console:
https://drive.google.com/open?id=1wfjR0ZGhZyZuQqOvvVcazmoyakQ7Odkd

kaiomatico, 2019-10-18:
Thank you for the quick response, I will try that on monday :)

kaiomatico, 2019-10-24:
Sorry for the late reply, I finally came around flashing the firmware. The log seems to not show any Alexa UDP searches as far as I can tell. I searched in the Alexa app by adding a new light->philips hue bridge v1. I also tried "search for new devices" by my voice. See the attached log.

Thank you so far for your effort. Maybe I am just making a user error? Can I help more like sniffing network traffic?
[log Fire TV Cube 2.txt](https://github.com/Aircoookie/WLED/files/3768988/log.Fire.TV.Cube.2.txt)

kaiomatico, 2019-11-17:
Yeah so I will just close this, it really seems that the fire TV cube behaves differently and does not send out proper search packets. I also can't get it to work with emulated wemo, ha-bridge etc... I just use sinric to access my ESP8266 for now... thank you anyway for this great project :)

closed 2019-11-17 by kaiomatico