wled/WLED #265: HTTP request API Hue not working (besides red)

Mattallmighty, 2019-10-17:
I am having issues with HTTP request API Hue not working
WLED version 0.8.5
sending http://192.168.1.102/win&HU=1 get red colour yeay

but when I send:
- http://192.168.1.102/win&HU=65535 stays red
- http://192.168.1.102/win&HU=200 stays red
- http://192.168.1.102/win&HU=2000 stays red

Aircoookie, 2019-10-17:
Hi!
I just tested this and I believe the hue api command is working correctly. The problem is just that the values you tested all correspond to red. The hue value cycles through all colors from 0-65535. Try 48000 and you should get blue, for example :)

closed 2019-10-29 by Mattallmighty

mscierski, 2023-05-19:
Hi, i need to control WLED by setting HSV/HSB by api, but its not working, eg:

192.168.0.169/winHU=48000&SA=250
or
192.168.0.169/winHU=48000
no change,
i tried hue range 0-65535 and 0-255 with no luck

i dont see Hue and saturation in JSON api