wled/WLED #266: Network Connection

ludeatude, 2019-10-17:
After installing wled on a NodeMCU. I went through the AP to change the network settings so it can connect with my network. Everytime I change it, it never connects. I have another NodeMCU that I went through this process with just fine but this one is not connecting.

Partizan7676, 2019-10-18:
It seems like firmware don't like spaces in network name.

ludeatude, 2019-10-18:
My SSID is only 1 name with no spaces.

brentbrooks70, 2019-10-21:
Check out #188 particularly [sansillusion's reply on 7/16](https://github.com/Aircoookie/WLED/issues/188#issuecomment-511837395) or possibbly [gsynthchris' reply on 9/8](https://github.com/Aircoookie/WLED/issues/188#issuecomment-529269679)
If neither of those apply to you can you provide a little more info, type of board, running from bin or compiling own, IDE version and core version of your board definition.

hristo-atanasov, 2019-10-21:
Hmmm .. I swear there is something wrong in the code. I have 15pcs NodeMCU V3 (two different models). I tried with 6 same boards. After many, many flashing I managed to save the settings only on two of them and they are now working. The strange thing is that I flashed these two particular boards may be 6-7 times each until somehow I managed to save the wi-fi and led strip settings. Another strange thing I saw is that the SSID and the password are saved, but after restart, it just does not connect to the wifi, but starts the AP again. When I connect to the AP I see that the SSID and Pass are there. :\ As all other boards are already flashed 3-4 times each, I feel a little tired of repeating same process again and again. What could cause such behavior?

PS: I tried the other model too. It is struggling the same way. Not a power issue - tried with 5V from 0.5A up to 30.0A

closed 2020-04-05 by stale[bot]