wled/WLED #269: Support for solid RGBW LED strips

Def3nder, 2019-10-18:
Hi Aircooookie,

thx fir this awesome project - it's amazing! 

I have quite a lot of non-digital LED strips attached to MagicHome Adapters re-flashed with Tasmota. 
Unfortunately beside a simple "color-change" there is no single effect available.

I think some of the effects in WLED would look good on single colour strips as well (like rainbow or breath etc.).

Do you think it would be possible to use WLED with an analogue 3 pin RGB strip (+ the common cathode) ?

The NeoPixelBus library does hide so much so that I could not find the point in the code where the colour(s) is passed to the PIN.

I would love to use WLED for the analogue RGB strips.

Greetings from Berlin,
Def3nder

Philje123, 2019-10-19:
Same for me too! I have 3 Magic home controllers with fixed RGB strips (non-addressable) running some custom MQTT code. But would love them all to be Wled. I really like it.

theshadowsvk, 2019-10-19:
Guys you can't have effects like rainbox on non-addressable strips. Imagine that on electricity level. You can only control each color on whole strip, not single diode without led-IC that controls it.
Just buy addressable strips and you are done, they are pretty cheap.
WLED project is only for addressable led strips, so the neopixelbus library, so the ws2812fx library.

Def3nder, 2019-10-19:
Sure, you can. Just define a WLED strip with a single pixel - all you can do with that can get done with an non-addressable strip, too.

Def3nder, 2019-10-19:
I think the way implementing this could bei tweaking the NeoPixelBus Library to offer another hardwsre type and add the logic to drive the three pins there...
What do you think?

theshadowsvk, 2019-10-19:
That's not the way it works. It can behave like single pixel cause it is analog. It lacks the IC which controls the chip, wled project,neopixelbus library etc sends data to this IC, it is data stream. And one IC cant control the whole led strip if you add one external like ws2811.

creativeretail, 2019-10-20:
This will do what you want:-

http://www.freetronics.com.au/freenfet3

Schematic / PCB files here:-

https://github.com/freetronics/FREENFET3

brentbrooks70, 2019-10-21:
@Def3nder this is a Duplicate of #58 Please see that issue for info on to make this work for you

Def3nder, 2019-10-22:
> @Def3nder this is a Duplicate of #58 Please see that issue for info on to make this work for you

@brentbrooks70 Thx - this looks exactly what I was looking for - I will try it and post my results here for reference.

Def3nder, 2019-10-28:
@Aircoookie : I saw you yesterday on Youtube in the Video-Chat with DrZzs and you mentioned the (analog) RGB strips and asked yourself "Why should anybody want to use analog RGB strips ?".
Normaly you're totally wright: the WS281x or SK6812 strips are way better but there is one big advantage of the old analog ones:

1) they are much brighter when you compare at price level
    you can get a 12V 5m RGBWW strip with 60 Pixel per meter 
   for the same price of a 5V 5m WS2812B strip and the later one
   is far less bright.

So, if you don't want an "effekt light" but rather want to have another lightsource,
this is more effective.

I have a 7m long Aluminium profile attached right below the ceiling in my daughter's room with 12V RGBWW strips in it and the light is as bright as 200W (old) halogen light.
She uses it for reading and additional lighting in gernal and 
I use it (controlled from Home Assistant) to display a color that tells her how much time is left in the morning: green = keep laying in the beed, yellow = time to stand-up, red = time to hurry up and begin with breakfast.

Right now I use Tasmota for this, but WLED is way more cooler !
So my goal is to get WLED running with analog 5-PIN RGBW strip, too.

I made the first tests to get (analog) solid color strips to work and basically is does work (with some restrictions - see below). I made the following insertions in **npbWrapper.h**:
  void Begin(NeoPixelType type, uint16_t countPixels)
  {
  [...]
	**_//init PWM pins
	pinMode(5, OUTPUT);
	pinMode(12, OUTPUT);
	pinMode(13, OUTPUT);
	pinMode(15, OUTPUT);
	analogWriteRange(255);
	analogWriteFreq(880);_**
    }
  }
  **_void SetRGBPWM(uint8_t r, uint8_t g, uint8_t b, uint8_t w)
  {
	  analogWrite(5, r);
	  analogWrite(12, g);
	  analogWrite(13, b);
	  analogWrite(15, w);
  }_**

  void SetPixelColor(uint16_t indexPixel, RgbColor color)
  {
	**_if (indexPixel == 0) //5050 strip is first pixel
	{
		SetRGBPWM(color.R, color.G, color.B, 0);
	}_**
[...]
  }

  void SetPixelColor(uint16_t indexPixel, RgbwColor color)
  {
	**_if (indexPixel == 0) //5050 strip is first pixel
	{
		SetRGBPWM(color.R, color.G, color.B, color.W);
	}_**
[...]
  }

With this the RGBW Color is set to the four ESP PINs, but:

 - the master brightness-slider does not work as expected.

You need to set master brightness to MAX and then choose the color with the RGB color wheel. This way you can get dimmed colors with effect "Solid Color".

But if you want to use other effects they will always run at 100% brightness no matter what is used in the master slider.

I will try to peek the NeoPixel Brightness and multiply it with the RGBW-Values next time ...

**Does anybody have other ideas ?**

raoulteeuwen, 2019-10-28:
> I use it (controlled from Home Assistant) to display a color that tells her how much time is left in the morning: green = keep laying in the beed, yellow = time to stand-up, red = time to hurry up and begin with breakfast.

:-)

Aircoookie, 2019-10-28:
@Def3nder that's a fair point with the higher brightness/lower cost of analog strips!

About the brightness slider not working, this is to be expected since that is handled by the NeoPixelBus driver library. We can easily implement the brightness ourselves though (with this it might even make sense to use 10bit instead of 8bit analog resolution, don't know if that makes a difference). First of all, make a new `byte bright;` var in the top of NpbWrapper.
In the SetRGBPWM then put:

```cpp
uint16_t rt = (r*bright) >> 8;
uint16_t gt = (g*bright) >> 8;
uint16_t bt = (b*bright) >> 8;
uint16_t wt = (w*bright) >> 8;
analogWrite(5, rt);
analogWrite(12, gt);
analogWrite(13, bt);
analogWrite(15, wt);
```

and in the setBrightness just add `bright = b`. 
Hope it works fine! (and I believe i will make analog LED PWM outputs available with a simple #define, making them officially supported)

Def3nder, 2019-11-06:
...I've added the code to a new branch of a fork:
https://github.com/Def3nder/WLED/commit/d1c289b709f0f9137180f8ff02ad5662f363b883

mason-xx, 2019-11-14:
This looks exactly what I need. I have some solid strips already installed and don't need some fancy effects but would love to utilize them via wled. Question is, can I flash this onto some of these controllers? https://github.com/arendst/Tasmota/wiki/MagicHome-with-ESP8285
I managed to compile it with the modification of @Def3nder and was able to flash it via nodemcu flasher. I see the wled hotspot, can connect to it but won't get a proper webif. Do i need to mod the code for the ESP8285 on this controller? These are pretty cheap and come in a nice package, so pretty ideal for the job. Sorry if this might a little oftopic but I'm kinda new to this.

Def3nder, 2019-11-15:
Hi @mason-xx , yes, I do have some of them in use, too.
They run perfectly with the modifies WLED.

There are two basic differenct versions out there: one with RGB only and another one with RGBW.

I have one with RGBW in use - to get the right colors from Alexa there are some additional changes necessary - if you want RGBW to drive, have a look here: https://github.com/Aircoookie/WLED/issues/315

Let me know if you have any problems.

mason-xx, 2019-11-15:
Thanks for the answer @Def3nder I wasn't able to get it up at all and went the esphome route. For the simple task of on/off and color change it works very well. and since I already had 3 of the RGB ones in use it won't justify a new purchase. I'll get back to WLED for the ws812b led ones.

gijsje, 2019-11-17:
I would love to see this included in the basic firmware as i have many analog strips and would love that they can work with the digital ones

wesmininger, 2019-11-21:
> Hi @mason-xx , yes, I do have some of them in use, too.
> I would suggest to get the one with RGBW and IR (U can use them for both RGB and RGBW) - I did get mines here:
> https://www.aliexpress.com/item/32884865866.html
> 
> They run perfectly with the modified WLED.
> 
> There are two basically different versions out there: one with RGB only and another one with RGBW.
> 
> I have some with RGBW in use - to get the right colors from Alexa there are some additional changes necessary - if you want RGBW to drive, have a look here: #315
> 
> Let me know if you have any problems.
Hi Def3nder!
Thanks for answering about the Magichome device, its what I have and I was looking for this info! 
When you say "they run perfectly with the modified WLED" what modification do you mean? do you refer to #315 you mention or some other modifications?

gijsje, 2019-11-23:
> ...I've added the code to a new branch of a fork:
> [Def3nder@d1c289b](https://github.com/Def3nder/WLED/commit/d1c289b709f0f9137180f8ff02ad5662f363b883)

Worked great for me using a H801 controller.
had to change the gpio output.
Thanks for this

Def3nder, 2019-11-25:
Hi @gijsje : this weekend I figured out how to use an IR Receiver with the H801.
Is this interesting for you, too ?

I used this remote: https://www.aliexpress.com/item/32987617960.html 
From the Controller I ripped out the IR Sensor - the PCB is labled so you know the right PINs for Vcc, GND and Signal of the sensor.

Then I connected the sensor to +3.3 , GND and GIOP0:
[image]

In the Code (in **NpbWrapper.h**) I disabled the button:  `#define BTNPIN -1`,
changed the IR PIN to 0: `#define IR_PIN  0`
and added the logic for the 40-key Remote in **wled20_ir.ino**.

The complete code is here: https://github.com/Def3nder/WLED

jihao, 2019-12-06:
I love this feature and will test it with my rgbw strip this weekend.
It will be great if this can be merge into aircoookie release.

GeorgeIoak, 2019-12-17:
@Aircoookie You had mentioned earlier in this thread that you were considering adding in @Def3nder code. I have another use case for it and that's with the Novastella 20W flood lights. They are RGBWW and although I haven't opened one up they appear to be very similar to the H801. I was able to flash them and control them with https://github.com/thehookup/E131-MQTT-RGBW-Lights/blob/master/E131RGBW_MQTT_Novostella.ino but that's just test code and doesn't take advantage of controlling WW and CW

Def3nder, 2019-12-18:
Hi @GeorgeIoak , 

according to https://blakadder.github.io/templates/novostella_20W_flood.html the board seems to use these PINs: RPIN: 4, GPIN: 12, BPIN: 14, WPIN: 5, W2PIN: 13.

We would propably need some testing but this should work.

I've just opened a pull request for analog RGBW(W) stripes.

Can you see what kind of ESP board is used ? 
It should be an ESP8266 or ESP8285.

As you flashed it, did you see the amount of flash memory ?

I could compile a version for you with some additional input ...

GeorgeIoak, 2019-12-18:
I know those pins are correct since the Arduino sketch I used has those defined. I haven't broken out CW and WW so I can't confirm that they are correct. [This site](https://esphome-configs.io/devices/novostella-20w/) has a little more information and link to the FCC report which shows a Tuya module but I don't recall that specific one and would have to dig up old notes to see if I have a reference to what looks like that. I could be wrong but in that FCC picture I see an external chip which I believe should be a flash chip and that would indicate a ESP8266 device.

I used tuya-convert to flash these and unfortunately I did not take note of what it reported and these are not easy to get into. Since I have OTA access to it I could try and flash some new code that presents the info on it's web page if you think that will help.

@Def3nder BTW, I didn't have much time but I cloned your repo and tried to compile on a Win10 box running VSCode w/ PIO. Unfortunately I was getting errors but that's happened to me with PIO so I need to dig into my setup and find why it didn't work. The bad thing about the code I'm running on this light is that it doesn't have a firmware update page so the only way to update it now is through Arduino OTA. I'm running some tests now on a Wemos D1 Mini to replicate what I have I have on the light so I can confirm the process will work before I attempt it on the light.

GeorgeIoak, 2019-12-18:
@Def3nder I flashed Tasmota on a 2nd flood light and it reports this:

```
ESP Chip Id | 8566815
Flash Chip Id | 0x1540C8
Flash Size | 2048kB
Program Flash Size | 1024kB
Program Size | 446kB
Free Program Space | 556kB
Free Memory | 30kB
```

tonyn0, 2019-12-31:
New to all of this, but love it! Are there are any inherent limitations for implementing _additional_ outputs using an ESP32?

tonyn0, 2020-01-01:
Baby steps! I just stumbled onto the fact that the ESP32 does _not_ support analogWrite, but found what is needed is ledcWrite. I'm working on this now...

Def3nder, 2020-01-06:
I've just implemented the ESP32 support for the solid RGB(W) stripes last weekend.
It uses the ledc function of the ESP32 framework and works quite well.

The downside is that the .bin grows a lot - uses roughly 80% of the program space with that change.
I will create a PR for the update the next days.

tonyn0, 2020-01-06:
@Def3nder Can you comment on "Are there are any inherent limitations for implementing additional outputs using an ESP32?"?

Aircoookie, 2020-01-06:
@tonyn0 ledc has 16 channels, so ESP32 should be able to control 4 analog RGBW or 5 analog RGB strips hardware wise.

@Def3nder wasn't the utilization at around 77% anyways previously?

Def3nder, 2020-01-06:
Hi @tonyn0 , on an ESP32 nearly all the PINs are PWM-capable and can be used for driving LEDs.
But for what reason do you need more than 5 PINs ?

@Aircoookie , I did not build the ESP32 variant before, I was only supprised to get something in the mid 80th -  this was with ledc and IR-Receive.
Some program space could be saved when de-selecting some of the IR protocols - there are very much IR-Remotes for "Air Conditioners" that use a lot of RAM / Flash mentioned in the header file, but at the first look the #defines for that are in the "IRremoteESP8266.h" and I didn't want to touch a file from an external library. Perhaps there is a way to change this defines without touching the library itself.

tonyn0, 2020-01-06:
@Aircoookie That's awesome.

@Def3nder I'm trying to replace an older (abandoned) PWM controller with four, RGBW outputs.

Def3nder, 2020-01-06:
@Aircoookie I just included the IRremoteESP8266 as "local library" under "src/dependencies" and changed all "SEND_...  defines" to false as well as the ones for Air Conditioners /they are all named for a common function in the header).

This shrinkes the .bin by 11k 😃 - isn't it worth including this library, too?
The actual version 2.7.2 has the advantage to support ESP32, too.

[IRremoteESP8266.h](https://github.com/Aircoookie/WLED/files/4026640/IRremoteESP8266.h.txt)

Aircoookie, 2020-01-06:
@Def3nder awesome! Great that they also added ESP32 support!
Submit a PR if you'd like :)

closed 2020-01-22 by Def3nder

Dreamoffice, 2020-04-07:
@GeorgeIoak 
i have also flashed the Flood lights with WLED but actually the CW-LED are not working. can you give me a hint where to look at?
`//this code is a modified version of https://github.com/Makuna/NeoPixelBus/issues/103
#ifndef NpbWrapper_h
#define NpbWrapper_h

//PIN CONFIGURATION
#ifndef LEDPIN
#define LEDPIN 2  //strip pin. Any for ESP32, gpio2 or 3 is recommended for ESP8266 (gpio2/3 are labeled D4/RX on NodeMCU and Wemos)
#endif
//#define USE_APA102  // Uncomment for using APA102 LEDs.
//#define USE_WS2801  // Uncomment for using WS2801 LEDs (make sure you have NeoPixelBus v2.5.6 or newer)
//#define USE_LPD8806 // Uncomment for using LPD8806
//#define USE_P9813   // Uncomment for using P9813 LEDs (make sure you have NeoPixelBus v2.5.8 or newer)
#define WLED_USE_ANALOG_LEDS //Uncomment for using "dumb" PWM controlled LEDs (see pins below, default R: gpio5, G: 12, B: 15, W: 13)
#define WLED_USE_H801 //H801 controller. Please uncomment #define WLED_USE_ANALOG_LEDS as well
#define WLED_USE_5CH_LEDS  //5 Channel H801 for cold and warm white

#ifndef BTNPIN
#define BTNPIN  0  //button pin. Needs to have pullup (gpio0 recommended)
#endif

#ifndef IR_PIN
#define IR_PIN  4  //infrared pin (-1 to disable)  MagicHome: 4, H801 Wifi: 0
#endif

#ifndef RLYPIN
#define RLYPIN 12  //pin for relay, will be set HIGH if LEDs are on (-1 to disable). Also usable for standby leds, triggers,...
#endif

#ifndef AUXPIN
#define AUXPIN -1  //debug auxiliary output pin (-1 to disable)
#endif

#ifndef RLYMDE
#define RLYMDE  1  //mode for relay, 0: LOW if LEDs are on 1: HIGH if LEDs are on
#endif

//END CONFIGURATION

#if defined(USE_APA102) || defined(USE_WS2801) || defined(USE_LPD8806) || defined(USE_P9813)
 #define CLKPIN 0
 #define DATAPIN 2
 #if BTNPIN == CLKPIN || BTNPIN == DATAPIN
  #undef BTNPIN   // Deactivate button pin if it conflicts with one of the APA102 pins.
 #endif
#endif

#ifdef WLED_USE_ANALOG_LEDS
  //PWM pins - PINs 15,13,12,14 (W2 = 04)are used with H801 Wifi LED Controller
  #ifdef WLED_USE_H801
    #define RPIN 04   //R pin for analog LED strip   
    #define GPIN 12   //G pin for analog LED strip
    #define BPIN 14   //B pin for analog LED strip
    #define WPIN 13   //W pin for analog LED strip 
    #define W2PIN 05  //W2 pin for analog LED strip
    #undef BTNPIN
    #undef IR_PIN
    #define IR_PIN  0 //infrared pin (-1 to disable)  MagicHome: 4, H801 Wifi: 0
  #else
  //PWM pins - PINs 5,12,13,15 are used with Magic Home LED Controller
    #define RPIN 5   //R pin for analog LED strip   
    #define GPIN 12   //G pin for analog LED strip
    #define BPIN 15   //B pin for analog LED strip
    #define WPIN 13   //W pin for analog LED strip 
  #endif
  #undef RLYPIN
  #define RLYPIN -1 //disable as pin 12 is used by analog LEDs
#endif

[... 216 lines cut ...]
  {
    switch (_type) {
      case NeoPixelType_Grb:  return _pGrb->Pixels();  break;
      case NeoPixelType_Grbw: return _pGrbw->Pixels(); break;
    }
    return 0;
  }

private:
  NeoPixelType _type;

  // have a member for every possible type
  NeoPixelBrightnessBus<PIXELFEATURE3,PIXELMETHOD>*  _pGrb;
  NeoPixelBrightnessBus<PIXELFEATURE4,PIXELMETHOD>* _pGrbw;

  void cleanup()
  {
    switch (_type) {
      case NeoPixelType_Grb:  delete _pGrb ; _pGrb  = NULL; break;
      case NeoPixelType_Grbw: delete _pGrbw; _pGrbw = NULL; break;
    }
  }
};
#endif
`

Dreamoffice, 2020-04-07:
@GeorgeIoak 
i have also flashed the Flood lights with WLED but actually the CW-LED are not working. can you give me a hint where to look at?

GeorgeIoak, 2020-04-07:
@Dreamoffice I still don't have WLED on these lights. You'll have to ask nicely to @Def3nder to see if he has time. The last time we spoke he was going to order one of the lights

roboraptor, 2021-02-21:
It is wery nice to jump into already going train like this! 

I just finally repaired an EGLO light with CWRGB leds.
Salvaged their bluetooth driver and replaced with esp8266 flashed with tasmota.
After few minutes fiddling with it through webUI and Alexa I was like "Damn! that is boring..." 
Wondering if something like WLEDs effects could take place and change the colors at itself instead of me still clicking on the colorwheel.

And here I am. 😄 Looking forward for this part to continue.