wled/WLED #270: Fade on/off

thespirit101, 2019-10-19:
WOW! - great job with the project and interface, love it and already planning my xmas lights, i have one question, all was going smoothly until yesterday, my strip no fades on and off, there is a delay of around 10 secs after cycling when it just used to switch on and off with the button press, have I inadvertently changed a setting? cant see anything obvious 

Cracking job Aircoookie - you rock!

Stay well 
C

brentbrooks70, 2019-10-21:
@thespirit101  What is the value listed in **Settings -> LED Preferences -> Transitions** _Transition Time_ and is _Crossfade_ checked?

thespirit101, 2019-10-22:
Thanks so much, you are of  course correct, no idea why it was suddenly checked, but thanks for the prompt, much appreciated

closed 2019-10-24 by Aircoookie