wled/WLED #271: ESP32 Wifi settings not saving

Boricuatec, 2019-10-21:
Thanks for making this project!  I was able to download the ESP32 binary from https://github.com/Aircoookie/WLED/releases  and upload so thats all good!  But when I hit the webserver and try to configure my SSID and password it wont connect to it.  I also noticed when I change the LED count from 30 to 300, it does not keep that setting.  If I restart the ESP32 it always goes back to 30.  Is there something special about getting the ESP32 to save those settings?  Is there a forum I could access to have others help me?

raoulteeuwen, 2019-10-21:
Can you link to the shop and device you're using? Just to check whether you have the right binary.

Boricuatec, 2019-10-21:
Hello @raoulteeuwen!

Here is the URL to the device i purchased: https://www.amazon.com/gp/product/B07Q576VWZ/ref=ppx_yo_dt_b_asin_title_o02_s00?ie=UTF8&psc=1

raoulteeuwen, 2019-10-21:
Thank you @Boricuatec , sorry for asking, just checking. Seems fine ;-).

 Do you have 2 ESP's or another controller you can try, to rule out a defective ESP / hardware?

chris89178, 2019-10-21:
Hi. I have the same problem. I verified this with two esp32 WROOM32. None of the changed settings are stored after reboot.

raoulteeuwen, 2019-10-21:
Mmmm, wondering whether the WROOM has some specific problem. Any chance you could try with another ESP, not a WROOM? Like https://nl.aliexpress.com/item/32647690484.html?spm=a2g0s.12269583.0.0.6136735667kiQe ?

How are you powering the ESP? While trying to find a possible problem, i found https://github.com/espressif/arduino-esp32/issues/2969 .

I also found https://esp32.com/viewtopic.php?f=19&t=4657 , hinting to some problem with (writing/)reading.

brentbrooks70, 2019-10-21:
@raoulteeuwen  It cannot be ESP-WROOM-32 specific as I have tested on two different ESP-WROOM-32 boards:
1. [Adafruit HUZZAH32 – ESP32 Feather Board](https://www.amazon.com/gp/product/B01NCRYHDL/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1)
2. [HiLetgo ESP-WROOM-32 ESP32 ESP-32S](https://www.amazon.com/gp/product/B0718T232Z/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1)

and both have no issues writing and reading WiFi Setup, LED Preferences, User Interface or Sync Interfaces settings across several reboots so it is not something specific to the WROOM chipset. Possibly the implementation of them by the manufacturer.
@raoulteeuwen could be on to something with his last comment from esp32.com were it seems some boards are being sent out with issues saving to EEPROM.

@Boricuatec I would try a different manufacturer of ESP32 and see if your issues persist. Can you try to run the eeprom_class example that should have been installed with the board definition? That is assuming you are using Arduino IDE.

sansillusion, 2019-10-21:
> found espressif/arduino-esp32#2969

I had similar experiences once, I had to format the flash chip for it to
start working correctly. Sharing just in case.

Digittt, 2019-10-21:
I'm on an ESP8266 and having the same issue.  Loaded onto another ESP8266 2 days ago and that worked no problem.

Aircoookie, 2019-10-22:
Thank you for all the reports!
I believe that this issue is due to an invalid SPIFFS configuration. You can try `esptool.py erase_flash`, it can often help with flash related issues.
ESP8266 users who build from source: Please make sure you select "Flash size: 4M (1M SPIFFS)"
ESP32 users: Seems like I made a mistake with the binary... I'll check and replace it if it is indeed faulty!

Boricuatec, 2019-10-24:
@Aircoookie Thanks for looking into this, when do you plan on updating this?  I would love to try it out and see if it works.

Aircoookie, 2019-10-24:
@Boricuatec just pushed the new code to master! Might not yet work well though, will release a binary tomorrow!

Boricuatec, 2019-10-24:
> [
> {
> "@type": "EmailMessage",
> "@type": "ViewAction",
> "target": "https://github.com/Aircoookie/WLED/issues/271?email_source=notifications\u0026email_token=ADTGT6Q4SC3KDY43TP4Z7G3QQINBTA5CNFSM4JDACP72YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOECGTJ4A#issuecomment-546125040",
> "url": "https://github.com/Aircoookie/WLED/issues/271?email_source=notifications\u0026email_token=ADTGT6Q4SC3KDY43TP4Z7G3QQINBTA5CNFSM4JDACP72YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOECGTJ4A#issuecomment-546125040",
> "name": "View Issue"
> },
> "description": "View this Issue on GitHub",
> "publisher": {
> "@type": "Organization",
> "name": "GitHub",
> "url": "https://github.com"
> }
> }
> ]

ThanksSent from my T-Mobile 4G LTE Device

Boricuatec, 2019-10-28:
Okay so it's partially working.  I can now set the WIFI and it will connect which is awesome.  But it the device gets powered down and restarted it, the wifi credentials are reset to default.

Aircoookie, 2019-10-28:
@Boricuatec ok, so it seems like there is no issue with WiFi then, but the saving of settings fails, probably due to a somehow invalid flash configuration. Did you try `esptool.py erase_flash`?

Boricuatec, 2019-10-29:
@Aircoookie It worked but I still think there is an issue.   I used the esptool to erase flash.  But I think that the save button in wifi settings is not really committing it to storage.  If I get it on wifi then go in to the security and updates then tell it to save it reboot then it will save it.  I could be completely wrong but I have flashed this device and another over and over again and it may have been the saving to storage that was the problem the whole time maybe???  They are working now so I am happy, I have to say I am totally loving your damn WLED!  Great work!  Also I saw your webcast with DRZZZs the other day!  Great job!!!

Aircoookie, 2019-10-30:
Tomorrow I will flash the binaries onto brand new ESPs and see if i'll be able to reproduce your findings!

ggrant111, 2019-11-07:
Was someone able to come up with a solution? I stalled the current .bin using ota programmer and having same issue. Can connect to AP but when I save settings for my local network and reboot everything is reset to defaults. Awesome program though! Great job!

Aircoookie, 2019-11-09:
Sorry for taking so long on this!
I couldn't reproduce any issue with the settings not saving yet...
What I reproduced though is another issue: https://github.com/espressif/arduino-esp32/issues/2501
Basically, in some cases the wifi only reconnects every second attempt. The auto reconnect will take care of this though and the ESP will connect after 5 minutes. Alternatively, power cycling or saving the wifi settings from AP once again will also work.

Here is a new binary, please test if it solves the saving issue. I'm a bit tapping in the dark with that one, so this will probably not solve the issue, but here you go:
https://drive.google.com/open?id=1n1zE7rhBjg5QrfP3Y3VKmudvXFgVnCYz

ggrant111, 2019-11-10:
> What I reproduced though is another issue: espressif/arduino-esp32#2501

Thank you I will try it out tomorrow.

Ukhauk, 2019-11-17:
For info I have the same issue WiFi credentials not saved after a reboot using the latest bin file
WLED_0.8.6_1911081_ESP32_ledpin16.bin board type is a wemos d1 r32

ggrant111, 2019-11-17:
what helped me was i reflashed the module and reloaded the bin file using
an OTA sketch on the board and since then has kept the settings.

Ukhauk, 2019-11-19:
I am trying to erase the ESP32 and re flash can anyone confirm I am doing the basics correctly an I missing anything extra when downloading a pre-compiled bin file ?

esptool.py --port /dev/ttyUSB0 --chip esp32 erase_flash

esptool.py --chip esp32 --port /dev/ttyUSB0 write_flash -z 0x1000 ~/Documents/Wled/WLED_0.8.6_ESP32.bin

Ukhauk, 2019-11-19:
Just an update in case anyone else having the same issues am I doing something wrong ?
I clear the ESP32 (ESP32DOWDQ6 revision 1) using the esp too commands 
esptool.py --port /dev/ttyUSB0 --chip esp32 erase_flash
uploading the standard esp8266 OTAWebUpdater program connects to my wifi 
using this to upload wifi credentials still don't save and connect after a re-boot
WLED_0.8.4_ESP32.bin
WLED_0.8.5_ESP32.bin
WLED_0.8.6_ESP32.bin

Aircoookie, 2019-11-19:
@Ukhauk did you try not setting a memory offset in esptool? Try this command:
`esptool.py --chip esp32 --port /dev/ttyUSB0 write_flash -z 0x0000 ~/Documents/Wled/WLED_0.8.6_ESP32.bin`

I'm not sure, but it might help :)

Ukhauk, 2019-11-20:
Hi Aircookie thanks for suggestion I have tried the above clearing the flash and esptool.py --chip esp32 --port /dev/ttyUSB0 write_flash -z 0x0000 ~/Documents/Wled/WLED_0.8.6_ESP32.bin
WLED doesn't run and I get the following error on the serial port rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT) flash read err, 1000 some google searches suggest this is a boot loader issue. I have 3 ESP8266 boards working without issue so its not critical to get this esp32 board working just thought it useful to report save other users.

I managed to get the OTAWebUpdater loaded by updating the ESP32 board manager to the latest revision. If I download WLED_0.8.6_ESP32.bin over the air I still can't save ths Wifi configuration as reported in this issue. I have the same issue with older version WLED_0.8.5_ESP32.bin

LArivee15, 2019-11-20:
I have the same issue on a wemos d1 mini ESP32 board. I have uploaded by USB and OTA using OTAWebUpdater and connected to WLED_AP but on power cycle/reboot the WiFi credentials have returned to default.  I have cleared the flash and re-uploaded with no success.  I too have tried with previous versions of WLED none of these seem to work.

I tried recompiling in platformio with hard-coded credentials but it was my first time with platformio and I am getting errors I cannot resolve at the moment.

andree182, 2019-12-09:
For the record, I have the same issue with "DOIT Esp32 DevKit v1", both 0.8.5 and 0.8.6 behave the same (only "remembers" settings until reset)

closed 2020-04-14 by stale[bot]