wled/WLED #273: WLED not receiving UDP from Hyperion

wardwygaerts, 2019-10-21:
I have WLED up and running, and it's working fine.
Now I wanted to send send my Hyperion config to WLED. I changed the config file with this:
```
"output" : "http://192.168.0.224:19446",
"protocol" : 0,
```

buy my leds connected to WLED are still showing the solid color which I've set. The top arrow is 'on' in the webinterface, so it should receive packets.
Do I need to specify a specific effect/setting?

brentbrooks70, 2019-10-21:
@wardwygaerts If you browse to **Settings -> Sync Interfaces -> Realtime** make sure _Receive UDP realtime:_ is checked.

DuschdrBabbe, 2019-10-22:
i think you should check your hypercon config again!

`// DEVICE CONFIGURATION 
	"device" :
	{
		"name"       : "WLED-Konfig",
		"type"       : "udpraw",
		"output"     : "192.168.178.69:19446",
		"rate"     : 1000000,
		"maxpacket"     : 1450,
		"protocol"     : 0,`
please remove "http:/" 

greetings

Pascal

wardwygaerts, 2019-10-22:
> i think you should check your hypercon config again!
> 
> `// DEVICE CONFIGURATION "device" : { "name" : "WLED-Konfig", "type" : "udpraw", "output" : "192.168.178.69:19446", "rate" : 1000000, "maxpacket" : 1450, "protocol" : 0,`
> please remove "http:/"
> 
> greetings
> 
> Pascal

hmm thanks for the info. Gonna check it this evening.
Do you know if it's possible to add multiple outputs? Because my Hyperion is running on my LibreElec Pi, and the leds of my TV are directly attached to this Pi. When I add the WLED ip in the output line, my leds behind my tv are not working anymore (obviously)

DuschdrBabbe, 2019-10-22:
Hi,

yes thats not a problem.. add a second config to your hyperion service like that: 

Edit your Service:
sudo nano /etc/systemd/system/hyperion.service

`ExecStart=/usr/bin/hyperiond /etc/hyperion/hyperion.config.json /etc/hyperion/wledtest.json /etc/hyperion/wled_tv.json`
after edditing your service you should reload it:
sudo systemctl -q daemon-reload
sudo systemctl restart hyperion.service

please note that you should use diferent ports in your config!

greetings

Pascal

wardwygaerts, 2019-10-22:
It's not working...
The Hyperion log reports "ssh in: 	Failed to open device('192.168.0.224:19446')"
telnet to 19446 is not responding, telnet to 80 is working.

Regarding a 2nd config:
In LibreElec, there is no 'hyperion.service' file, but these 2 files:
- hyperiond.sh

`#!/bin/sh
LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/storage/hyperion/bin /storage/hyperion/bin/hyperiond "$@"`
- autostart.sh
`/storage/hyperion/bin/hyperiond.sh /storage/.config/hyperion.config-default.json > /storage/logfiles/hyperion.log 2>&1 &`

adding a 2nd config before the '>' is not working.

wha do you mean with using different ports? I removed all forwarding configs.

DuschdrBabbe, 2019-10-23:
If you start a second hyperion instance you need to adjust the json boblight and protoserver port in your second config file... search for this entrys in my case row 120
`	// JSON SERVER CONFIG
	"jsonServer" : 
	{
		"port" : 19448
	},

	// PROTO SERVER CONFIG
	"protoServer" : 
	{
		"port" : 19449
	},

	// BOBLIGHT SERVER CONFIG
	"boblightServer" : 
	{
		"port" : 19450,
		"priority" : 900
	},
`
you can not use any ports
there are already occupied... 
in my case i have used the 3 next ports service 
the first config  use port 19444-19446 and the second 19448-19450.

wardwygaerts, 2019-10-24:
Thanks for the help. I finally managed to get it work! My leds attached to the nodeMCU are finally getting the correct colors from Kodi.

I noticed that when I manuammy change colors (and sometimes during effects of hyperion), I have some flickering, but this is another issue (already mentioned here: https://github.com/Aircoookie/WLED/issues/230)

closed 2019-10-24 by wardwygaerts