wled/WLED #274: WLED Alexa Integration

MalagaInternet, 2019-10-22:
Hi, 

First of all thank you very much for the work. It is wonderful.

I use Wemos D1 Mini with WLED 0.85 and I have issues with alexa responses.

When I execute a voice command ( on / off / change color ) the action is done but Alexa always say "Device not respond. Check Network configuration and electrical power supply"

I wait you can take a look at the alexa integration. ;-)

Def3nder, 2019-10-23:
I had equipped two D1 mini's with WLED and one of them is acting "okay" the other one has the same problems that you describe.

The reason of the error has been the power supply - the 2nd one had a cheap one from China and this interfered with the WLAN.

Maybe you try to change the power supply and try again ?

MalagaInternet, 2019-10-23:
Hi def3nder

Thank you very much for your help and sorry for my bad English.

I think that in my case it is not a hardware problem or at least it is not related to the power supply because I tried to power the Wemos from various energy sources:

1.- From pin 5v and pin GND with an MB102 power module
2.- From microusb with Rpi3 power supply.
3.- From microusb with USB PC

In all cases the behavior is the same. The action is done, but Alexa always says "The device does not respond. Verify the network settings and the power supply."

Have you tried both Wemos D1 with the same power supply?
If the power supply is the problem, both Wemos D1 should work with the correct power supply. 

Is that so?

If You have a Wemos D1 Mini that works correctly, the software problem could be ruled out and maybe the problem is in my Wemos D1 Mini.

Now I only have a Wemos to try. On Friday I will have more units and I will inform you

matthiasnowak, 2019-10-23:
I will +1 on this.
I'm using a custom made ESP8266 board, freshly flashed 0.8.5 and just discovered by alexa. Power supply is steady at 3.3V.
Same problem as @MalagaInternet , I can say the command "turn on/off", it does the action without delay, but alexa responds with "not available".

I might get to test it with an older firmware tomorrow.

MalagaInternet, 2019-10-31:
I have done tests programming 3 Wemos D1 Mini and 3 NodemCU with .bin of wled 0.8.4, 0.8.5 and 0.8.6
I Have tried power them with different power supplies.
I have also tried compiling from the code but in all cases the issue persists.

Definitely alexa integration does not work properly.

MalagaInternet, 2019-10-31:
Hi @Aircoookie 

Thanks for your reply. I use "Amazon echo dot 3".

raptusis, 2019-11-02:
Yes, I have the Same problem with Echo Dot 3 gen and NodemCU

(Thanks for this project and Sorry for my english)

Aircoookie, 2019-11-02:
See Aircoookie/Espalexa#88! Someone found a fix, I will release a new espalexa version which will fix this issue as well!

raptusis, 2019-11-03:
Thank you Aircookie , I think your project is essentially the best and the most complete.

MalagaInternet, 2019-11-03:
It´s fantastic news My mom could control the Christmas lights !!

Thanks @Aircoookie

MalagaInternet, 2019-11-03:
Confirmed. Issue has been fixed with https://github.com/Aircoookie/Espalexa/pull/88

closed 2019-11-08 by MalagaInternet