wled/WLED #275: Reconnect WiFi

lufton, 2019-10-23:
I have WLED installed on nodeMCU. The issue I faced with is it wouldn't try to reconnect to WiFi.
We had couple power loss recently. So both my router and nodeMCU were powered on the same time, but router has longer boot time, so WLED wasn't able to connect to network. I don't know if it actually tried couple times, but eventually it gave up before router actually completed booting process. And it just start WLAN-AP provisioning. After resetting nodeMCU it successfully joined my network.
Is there some sort of setting to tell WLED to keep trying to connect instead of starting AP?

raoulteeuwen, 2019-10-23:
Does https://github.com/Aircoookie/WLED/issues/261 answer your question?

lufton, 2019-10-23:
> Does #261 answer your question?

Yes, definitely! Thanks!

closed 2019-10-23 by lufton