wled/WLED #276: Firefox missing slider hints in web app

tomlut, 2019-10-24:
The colour strips under the sliders that identify their purpose are not visible in firefox . Works fine in chrome and safari. But looks like this in firefox 69.0.3:

[image]

Aircoookie, 2019-10-24:
Hi, thank you for the report :)
Seems like Firefox updated their rendering recently, the way these sliders were displayed always seemed a bit off.
I likely won't be fixing this issue as the classic UI will be replaced by a new unified UI in the next major version.

tomlut, 2019-10-25:
It's not a major issue as I mostly use the iOS app. As you have plans to fix this in the next release I'll close the issue.

Thanks for this excellent software and support.

closed 2019-10-25 by tomlut

HimbeersaftLP, 2019-10-25:
Would you still accept a PR that fixed the issue?