wled/WLED #277: E1.31 seems to only listen to one universe

CaffeinatedFunctionality, 2019-10-24:
I'm trying to play with WLED's multicast and XLights. Admittedly, this could be my lack of knowledge here. I'm running 3 node MCUs One running about 400 lights the other running about 500 lights and one running 260 or so. Anyways, the way I understand E1.31 is each universe will only allow 510 channels which is about 170 leds. I'm expecting that I wouldn't need a nodemcu running WLED every 170 leds. It allows you to put in the starting channel and it seems to only listen to that channel and not any more than that. Now for Unicast that seemed to work correctly, however once it's on it loses WIFI signal and I guess resets the chip. I'm not sure if that's a memory issue, but what is the most leds these things can run E1.31 on a single nodeMcu. Is this software related or an issue with the nodeMcu? Thank you for your hard work on this software, it truly is the best I have seen.

Aircoookie, 2019-10-24:
Hi! The theoretical maximum WLED is programmed to handle in E1.31 mode is 680 LEDs (4 universes). It should listen to the starting universe as well as the subsequent 3 universes.

How much works in practice depends on a few factors, such as how good the wifi connection is and the frame rate at which xLights is sending data. Anything over 200Kbps might crash the ESP. ESP32 fares a little better in that regard. This value is quite easily reached, unfortunately.
200Kbps = 25KB/sec = assuming 50fps, 500bytes per frame = a little over 150 LEDs. YMMV :)

Thank you! I am truly happy that you like WLED, and i'll do my best to improve it in the future!

hristo-atanasov, 2019-12-02:
I have he same problem on all 9pcs. ESP8266 NodeMCUs. Only the first (from 2) universe is working. The resting one is not. The wi-fi connection is perfect and the router allows 450Mbit connections ..

closed 2020-04-07 by stale[bot]