wled/WLED #278: Custom & Favorite Colors

jakeyjava, 2019-10-24:
Hi Aircookie,

first of all, I love this project.  I have permanent christmas lights on my house and I've decided to swap out the supplied controllers and use WLED because of how amazing they are.  I have zero coding know-how so I was so pleased to find a project this cool.

One thing that I am sad to leave behind from the company-supplied app is the idea of custom color palettes.   I saw in another issue you had a release coming soonish that might address that but I got lost on what 'via Json' meant.  Does that mean it won't be in the mobile or desktop UI?  

Have you ever given thought to adding an R/G/B + Brightness slider in addition to the wheel?  This would be super helpful to make repeatable colors absent a 'color favorites' feature.

Thanks again so much, this has been such a fun experience as my first little project.  I didn't even think I'd get past setting up the NodeMCU. 

Thanks!

Aircoookie, 2019-10-24:
Hi,

welcome to WLED! It makes me truly happy that you like my project this much!
No, custom color palettes will of course be available via the UI! (although it might take me a bit to add it, first need to create the API, so JSON will likely work before the UI components)

I personally don't really like RGB sliders. They are the basis of the old classic UI (because I didn't know about the awesome iro.js colorwheel at the time) and it is way harder to pick a nice looking color or get the exact hue you are looking for IMHO. But I might consider it as an alternative option to the wheel soon.
For getting repeatable colors, I would recommend the presets feature! It let's you save color/effect combinations to one of the 16 save slots!

Let me know if you need anything!

jakeyjava, 2019-10-24:
wow, thanks for the quick response!  I've been playing all day.

jakeyjava, 2019-10-25:
One more question!  Sorry I’m being a pest. 

I’m trying to figure out how to get the lights to turn on and off at a certain time each day. I’m playing in the time and macros screen and I just can’t figure out how it works.

Aircoookie, 2019-10-25:
No worries :)
Let's use #246 for further discussion on custom palettes by the way!

Try this: (here the light would turn on at 6pm and off at 10pm)
[image]

jakeyjava, 2019-10-25:
Thank you so much. It's so much simpler than what I was trying.  

I'll follow #246 for more updates on the palettes.  

Thanks again for everything you do.  I hooked my lights up to my little breadboard contraption last night and I had two neighbors with permanent lights come over and ask how I got such complex patterns running.

bitterdog, 2019-12-13:
PERFECT! Thank you

closed 2020-04-18 by stale[bot]