wled/WLED #279: QuinLED-Dig-Uno temperature sensor support

intermittech, 2019-10-24:
I'm the creator of the QuinLED-Dig-Uno boards which work excellently with either a ESP8266 or ESP32 to run Digital LEDs using WLED, for more info see: https://quinled.info/2018/09/15/quinled-dig-uno/

One of the features on the board is that it has a spot where people can add a DS18B20 temperature sensors and the 4.7K resistor that goes with it. That works perfectly with for instance ESPhome and other firmware but WLED doesn't have an option to do a read out of it.

**Feature suggestion: Add ability to read-out temperature sensor in WLED (in app and MQTT)**

I think adding this could be useful for people using controllers inside of an enclosure. Since the sensor is just on the board and not isolation it's less suited to measure room temperature but it does show temperature changes really well.

I have received this request on a few of my videos and in my Discord. :)

Aircoookie, 2019-10-24:
Alright, I believe you have convinced me to add this soon!
It is correct that quite a few users have requested better sensor support and my stance was that sensors are outside the scope of the project. But given that many users wish for support for various more sensor types (temperature/distance/motion/light/more buttons), I believe it would make sense to add it! It is also a good argument that running a second ESP just for sensor tasks in uneconomical if the WLED ESP can 

However, this will likely take a bit of time as I would like to have an easily usable API for adding additional equipment and sensors to WLED. If I just add support for DS18B20, people using DHT11 will want me to support that as well, and so on. That would take too much of my time as well as bloat the codebase with something that 99% of users don't need (looking at you Cronixie support, you'll be banished from the core and become a usermod soon).
So my vision for this would be to have the user drop in a bit of sensor specific code and have the API create the framework (readout in the web UI, corresponding MQTT topics) automatically :)

Mariu86, 2019-10-25:
Interested in that ! Very good Aircoookie! Thanks again! :D

400killer, 2019-10-29:
This is what I did to get the temperature sensor on the QuinLED-Dig-Uno (DS18B20) working for MQTT. I am no programmer, and my first post on github! I am just a guy trying to learn and thought I would post what I did to get it working in the meantime.  I got it working on my 8266 D1 mini and NodeMCU on wled 0.8.5, to report back to MQTT on the "topic you program in the UI/temperature". Hope it works well enough to get other going for those that use MQTT for home automation. I use it to send alarms if the board gets hot.

1) Paste the code in the wled00.txt in the top of wled00.ino, This calls the libraries needed and declares variables. I couldn't get them to function corretly in the usermod file.
2) Paste the code in wled06_usermod.text to replace the code in wled06_usermod.ino. This portion starts the sensor service on boot, and reports the temperature to serial and MQTT every 60 seconds.

It will work for 0.8.6, you just need to paste the codes for the usermod file into the appropriate function declarations. The function declaration names changed between versions.

Last and not least, Thank you so much Quindor and Aircoookie for the immense amount of work you guys put in! The wled firmware is simply amazing, and I don't want to think about the amount of time yo have in it. And the Quindor boards are awesome. I stuff these in remote locations in the house/attic. I love the power management, and protection they provide to give me piece of mind in these locations. Hats off on the design from an electrical engineer.

[wled00.txt](https://github.com/Aircoookie/WLED/files/3781685/wled00.txt)
[wled06_usermod.txt](https://github.com/Aircoookie/WLED/files/3781686/wled06_usermod.txt)
[image]

intermittech, 2019-10-30:
@400killer wow, that's an awesome hack! A good intermediate until @Aircoookie has the time to work on his sensor integration framework. :)

400killer, 2019-11-01:
Thanks, I need to work on it. I need to add a check if the mqtt is connected before sending.
It works okay, but crashes the controller if the mqtt is not setup. You have to be fast between reboots, I have it publish the temp every minute. so you have a minute between reboot, lol.
Right after a reboot, use the app to change info to your WiFi as normal, save and reboot.
Setup the app to connect to it.
Right after a reboot occurs work to input your mqtt setting.
It will stop resetting. I will post up the code for people to insert on my gethub when I get the check for mqtt working.

Aircoookie, 2019-11-01:
@400killer here is the check I use to find out if MQTT is connected. Hope it helps you as well :)
`if (mqtt == nullptr || !mqtt->connected()) return;`

400killer, 2019-11-07:
Thanks for helping and pointing me in the right direction Aircoookie. I have added the check and added the two files in my github for 0.8.6.

https://github.com/400killer/WLED-Quinled-board-MQTT

huggy-d1, 2020-01-06:
I wonder if your mod could be configured so it did not require changes to wled00.ino?  I think so.  I would be willing to submit something for testing if interested.

I ask because I'm working on a usermod for the QuinLED Dig-Uno for PIR motion sensor control of the LED strip.

It would only require changes to the usermod.ino, but would need to cooperate with your mod.  I have something sketched up that you might consider.

My usermod also includes your usermod code since they are on the same board.

400killer, 2020-01-06:
Absolutely! Also it is listed in this repo under the usermod folder, it is under view code section. I was not able to get it to initialize without putting it in the wled00 file. I tried to use the links in the wled06 file for iniatializing, bit couldn't get it to work. Let me know, I will help test. Simplier is always better. Thanks for the help. Excited to see the pir integration.

huggy-d1, 2020-01-08:
@400killer According to @Aircoookie you can insert everything that is inserted in wled00 into wled06_usermod before the first function call.

Also, I posted my #conditional #compilation QuinLED Dig-Uno in another thread a few minutes ago.  I would appreciate it if you would try it out with your code sometime and see if it works for your setup.

After the #conditionals, I have this in the usermod I'm working on:

`/*============================================`
` * QuinLED DigUno Temp Sensor userSetup
 */`
`  //Initiating code for QuinLED Dig-Uno temp sensor`
`  #ifdef USERMOD_QUINLED_DIG_UNO_TEMP_MQTT`
`    #include <DallasTemperature.h>`
```
    OneWire oneWire(QUINLED_DIG_UNO_TEMP); 
    DallasTemperature sensors(&oneWire);
    long tempTimer = millis();
    long lastTempMeasured = 0;
```
`    //Uncomment Celsius if that is your preferred temperature scale`
`    //#define Celsius`
`  #endif`
`/*============================================*/`

It is your code and will have proper attribution before it's posted.

austwhite, 2020-01-11:
Just watching this as I am interested.  QuinLED combined with WLED makes one of the best open source LED controllers available.  Nothing else comes close in my experience.
Thanks for all the brilliant work you guys do.

grandfx, 2020-02-29:
Any Progress on this thread?  Looks like it slowed down in January and haven't seen any updates.  I would love to see this as a WLED feature.

theoracle09, 2020-03-28:
I also just want to chime in and say I'm using WLED on a dig-uno board, and would love to have native support for the temp sensor. I already have temp and humidity sensors near the controllers, so I'm not interested in hacking and changing files. I'll patiently wait for a solid solution. I'm also available for testing if anyone needs it. Thank you Aircookie, and Quin for what you both do and have provided the community!

ricargr, 2020-06-03:
Hi!!
Is this request still in progress?
Regards

Elkropac, 2020-06-14:
> It is correct that quite a few users have requested better sensor support and my stance was that sensors are outside the scope of the project. But given that many users wish for support for various [...]

It would be great, if wled firmware was converted to set of components for esphome (webserver component, light component, ...), then you should not care about sensors anymore ;)

But i guess, it would be very hard to do.

Aircoookie, 2020-06-15:
@ricargr in fact it is done! It is a v2 usermod capable of displaying the temperature in the Info section of the web UI! See https://github.com/Aircoookie/WLED/tree/master/usermods/Temperature

@Elkropac that would be a bit difficult indeed, just because of limited resources on ESP8266. WLED and ESPhome also serve different core purposes. In any case, I believe it is a tall order to request both control of hundreds of individually addressable LEDs and support for a lot of different sensors from an ESP8266 :)

ricargr, 2020-06-15:
> @ricargr in fact it is done! It is a v2 usermod capable of displaying the temperature in the Info section of the web UI! See https://github.com/Aircoookie/WLED/tree/master/usermods/Temperature
> 
Thank you and great job with WLED

400killer, 2020-06-15:
@Aircoookie thanks so much for including this in the repository! I am a little sad I didn't get any credit in your post, since it is my original code (see above). But honestly, I'm good, as long as it's their for people to use. Thanks again for all your hard work.  I currently use 6 controllers on my house, and one on my car, lol. Your work and commitment to keep this project going is appreciated by thousands!

Legsmaniac, 2020-06-15:
> Your work and commitment to keep this project going is appreciated by thousands!

Hear, hear! 👍

Aircoookie, 2020-06-15:
@400killer Thank you :) I am sorry for not including attribution to you in the usermod, I did not know this was your code, since I modified it from @srg74 's [usermod](https://github.com/Aircoookie/WLED/tree/master/usermods/QuinLED_Dig_Uno_Temp_MQTT). But it seems like his code is partly based on yours, so I'll add the credit for you both!

400killer, 2020-06-15:
@Aircoookie Thanks man. Yeah he put some work into it too, and should be credited. I was just a little bummed since I have never contributed to anything on GitHub before. I'm just learning how GitHub and coding works. I'm super stoked, thanks so much!

400killer, 2020-06-15:
> @400killer According to @Aircoookie you can insert everything that is inserted in wled00 into wled06_usermod before the first function call.
>
> Also, I posted my #conditional #compilation QuinLED [...]
@huggy-d1 sorry man, I just saw this and realized I totally spaced trying it out for you. My bad...half a year later.

ricargr, 2020-08-07:
Hi
Here having problems  for esp32 with the temperature module (no problems with esp8266). Once compiled, the leds doesnt respond. If compiling without the temp module all ok.
Anyone experiencing same issues?

400killer, 2020-08-10:
Sorry, I have only run it on the esp8266

ricargr, 2020-08-10:
Thanks for your reply. The problem was I was compiling with arduino. With platformIO it works well.

blazoncek, 2021-09-13:
I think we can close this since temperature usermod is now standard feature in WLED. Do you agree?

nibblerrick, 2021-09-13:
Standardfeature is there are usermods availible and one is a temperature usermod. Or is temperature in not-usermod-WLED included?

blazoncek, 2021-09-13:
usermods/Temperature
it is not included in default builds.

closed 2021-10-03 by blazoncek