wled/WLED #283: Sync Issues

ludeatude, 2019-10-25:
My lights are not in sync when I turn it on in the app. Any thoughts?

Aircoookie, 2019-10-26:
Thanks for the report!
Did this happen after you updated to 0.8.6?

ludeatude, 2019-10-26:
Did not know there was an update. Can I do this OTA with a wemos d1 mini?

ludeatude, 2019-10-26:
Still the same issue. Each controller is connected 65 mbps too.

Aircoookie, 2019-10-26:
Ok, thanks! I believe that the default configuration might somehow be wrong for you. Can you take a look at WLED sync section under sync settings? Most of the checkboxes should be ticked.

ludeatude, 2019-10-27:
I haven't deselected anything in that section that was already selected. Anything I should make sure IS selected?

ludeatude, 2019-10-27:
Receive, brightness, color, effects, send Phillip's Hue change notification, udp realtime. This is what is checked.

Aircoookie, 2019-10-27:
Thank you! That doesn't sound like it's the source of the issue, but to make sure you could enable Sync on direct change (this is what is normally toggled by the slider in the UI). Also please check that both UDP ports are the same (default is 21324). Lastly, UDP multicast might be disabled in your router. You can look whether it has a setting for that. Some models also call it UPnP.
I hope we can get sync to work for you!

frenck, 2019-10-28:
@Aircoookie I've noticed the same now. It seems to be limited to the JSON API. If an effect change is made through the JSON API, the change event is not broadcasted.

However, if an additional change is made after that (e.g, change of primary color), it gets broadcasted as part of that extra change.

So judging by that, I would say there is a event missing...

Aircoookie, 2019-10-29:
@frenck Good observation! I would have never cracked down this missing event without your detailed description! Should be fixed in master now!

frenck, 2019-10-29:
Your welcome m8, glad you've got it resolved.

frenck, 2019-10-30:
@ludeatude Wait for the next release? 😉 
Or build it from source yourself.

Aircoookie, 2019-10-30:
@ludeatude is your sync limited to the JSON API FX issue @frenck detailed, or does it not work at all even when only doing something simple, like setting brightness in the app/web UI?

ludeatude, 2019-10-30:
It works but the 2 controllers get out of sync very easy.

closed 2020-04-05 by stale[bot]