wled/WLED #284: Hardware factory reset

mehranviva, 2019-10-26:


mehranviva, 2019-10-26:
Hi first thanks to your good project.
How i can add a hardware factory reset in this project? Maybe i forgot password and cant program chip again
And next question is there any security to encrypt code from read ?
Thx alot

mehranviva, 2019-10-26:
Hi first thanks to your good project.
How i can add a hardware factory reset in this project? Maybe i forgot password and cant program chip again
And next question is there any security to encrypt code from read ?
And last is ,Can the output pin be logical not?
Thx alot

Aircoookie, 2019-10-26:
Hi and welcome!
If you connect a pushbutton between GND and GPIO0, you can use that to turn on and off the LEDs. You can also hold it for 6 seconds to restore access point credentials to defaults! Otherwise, you can always reflash the software via USB.
The flash (binary program and user config) can be read out by the esptool software. So unfortunately, no security there but this shouldn't be an issue for use in homes and personal networks.
You mean whether the output logical level could be inverted? Yes, you can do this now. Update Neopixelbus to 2.5.3! Try changing `NeoEsp8266Uart1Ws2813Method` to `NeoEsp8266Uart1Ws2813InvertedMethod` in NpbWrapper.h
I hope it will help :)

mehranviva, 2019-10-26:
Thank you very much for the full explanation
The thing that came to my mind for security is what do you think of the device locking through Mac Adress? Where can I write this code that does not disturb performance?

closed 2020-04-05 by stale[bot]