wled/WLED #285: on/off switch support

DismissedGuy, 2019-10-26:
I know WLED has support for push buttons to trigger macros and cycle through effects, but I was wondering if it'd be possible to use an on/off switch to turn the strip on or off?

Ideally, the application would set the brightness to 0 when the switch is turned off and restore the brightness when it's turned back on. However, the switch should override any requests from other applications to turn the strip back on, since that would ruin the purpose of the switch.

I know I can code in basic support in `wled06_usermod.ino`, but would I be able to override requests from clients using this method? If not, can anyone point me into the right direction as to which file I should instead take a look at? I still have to familiarize myself more with the project.

I can't wire my switch in series with the power supply, since that would power off my esp8266 and prevent it from saving its last state, causing it to use the default when powering back up.
Simple solution would be to just interrupt the data signal with my switch, but I'm uncertain whether this approach would work.

Any ideas, answers or pointers into the right direction would be greatly appreciated.

Aircoookie, 2019-10-26:
Hi and welcome!

I might add switch support in the future :) That would still be overridable by other interfaces though, since the switch is just one method of controlling.

For your use case the simplest way would be running a separate 5V from your PSU to the ESP that is always on and interrupt the 5V of the LED strip with the switch (make sure it can handle the current!)

Have fun with WLED and let me now if you have any other questions! :)

DismissedGuy, 2019-10-26:
Thank you for the quick reply!

The "problem" I think that arises when you don't allow overriding with a switch is that it allows the LEDs to be on while the switch is turned off, which would require flicking it on and off again to physically turn it off again. The physical switch would get out of sync with the program's virtual "switch".

Other than that, I am already running two separate lines from my power supply to my ESP and LED strip, but I assume disconnecting the strip's 5V wouldn't damage its ICs then? I've found the datasheet for the WS2812B to be quite vague, so I'm not sure what to expect from its behaviour.

Aircoookie, 2019-10-26:
I understand where you are coming from!

Disconnecting 5V should be safe, from what I've read only GND needs to be connected permanently. If you want to be extra careful you could use a 2 phase switch and disconnect data as well.

DismissedGuy, 2019-10-30:
I just tried out the method you suggested, it seems to work perfectly fine other than some visual glitches when toggling the switch which I assume are caused by the esp continuing to send data to the strip while it's off, causing it to only partially receive the instructions when the IC chain is powered back up?

Anyway, these glitches are very minor and I am completely fine with the current solution, so I am just going to close this issue since I don't think there is a need for software support.

Thank you for helping me out, and keep up the good work! :)

closed 2019-10-30 by DismissedGuy

Adrianormrs, 2021-03-09:
Hi! Any news about the switch support? I've been trying to make ir work with a common wall switch to toggle on/off but working as a 3 way switch with other control inputs such as Alexa, but I couldn't make it work so far.

ThomasCr, 2022-10-15:
I would also be interested. I would mount the LEDs on the kitchen sink. There is currently a Switch outside which I would use as a toggle Switch. When I switch on it should run it in a defined program/brightness/color. This would be awesome and rise the usecase!

ThomasCr, 2022-10-15:
Okay, found #1053