wled/WLED #286: Problem with latest update WLED_0.8.6

raufis27, 2019-10-26:
NodeMCU running 0.8.5. After updating to 0.8.6 doesn't boot. Had to roll back to 0.85.

CaffeinatedFunctionality, 2019-10-26:
Mine booted up but it took a rediculously long time. Some wifi settings were adjusted in this release which may account for this. 

⁣Sent from BlueMail ​

Aircoookie, 2019-10-26:
Thanks for the reports! That doesn't sound too good...
Did you use the binaries or compiled from source?
If you compiled yourself, please try enabling "WLED_DEBUG" in wled00.ino :)

raufis27, 2019-10-26:
I did use binary file. I did wait about 6 minutes, I think it was in boot loop because lights were blinking every 10 seconds. Then I flashed it with OTA firmware and uploaded 0.8.6 again. But it restated with OTA firmware. So I gave up and uploaded 0.8.5.

bradmck, 2019-10-26:
I had the exact same thing happen.  It was rebooting and the lights would go off and back on every ~10 seconds.  I enabled debug and could see it do a watchdog reset after going into STA mode.  I downgraded to 0.8.5 but it didn't help.  I finally switched Arduino IDE to "Erase Flash: All Flash Contents" and it went back to AP mode when I flashed.  I then flashed 0.8.6 and it booted up fine.  All was good until I went in and enabled MQTT and E1.31   It went back to rebooting.  I went back to 0.8.5 (erasing all flash) and it's fine again.  I didn't have enough time to try enabling MQTT and E1.31 one at a time to see if that helped.

I'm running a NodeMCU with 96 SK6812s (two 48 LED strips both hooked to the same pin)

Thank you for this great project!

eliobou, 2019-10-26:
Same problem for me, rolled back to 0.8.5

Aircoookie, 2019-10-26:
Sorry for all the trouble!
I believe this is due to an issue with mDNS in LwIP 2 I believed to have fixed, but that seems to still occur with a certain combination of settings/flash contents.
You can try the `esp8266_lwip1.bin` I just uploaded to the releases page or compire with lwip 1.4 Higher Bandwidth and everything should work (except Homeassistant autodiscovery, which crashes the MQTT connection in 1.4, that's why I defaulted to 2)

I've enabled both MQTT and E1.31 multicast and can't reproduce the issue. So it must be due to some setting or flash content. I'll continue attempting to find the cause.

marc-gist, 2019-10-27:
I get it to boot fine; however, notice that mDNS no longer works (can't access via <name>.local :(

marc-gist, 2019-10-27:
Futher testing, if I compile with lwip 2, boot loops but sometimes goes stable and mDNS works again.  But lwip 1.4 is more stable, but no mDNS (or auto discover via app) :(

marc-gist, 2019-10-27:
More testing results.  Issues were noticed on a Wemos D1 Mini v2.
Flashed a Wemos D1 Mini v3 with lwip 2 HB set and it works fine! :)  so might be something with different esp chips :(

eliobou, 2019-10-27:
I'm using a Wemos D1 Mini v2 too

marc-gist, 2019-10-27:
> I'm using a Wemos D1 Mini v2 too

So that's clearly the issue. My D1 Mini v3 (have 4 running) work fine.  Was testing custom builds on a v2 board and it fails with new code and LWIP2

raufis27, 2019-10-27:
I used esp8266_lwip1.bin to upgrade NodeMCU. It worked but MQTT didn't work, or it works but topics changed ( have to confirm). Tried to go back to 0.8.5 but can't, it is stack in boot loop. Probably have to erase flash. So I used different NodeMCU and flashed it with 0.8.5.

marc-gist, 2019-10-27:
I have a version work on my D1 Mini v2 now... compiled with setting board to "NodeMCU 0.9 ESP-12" lwip 1.4 high bandwidth.  Flashed, with full erase set.

bradmck, 2019-10-28:
I'm not running on actual D1 Minis, but they're clones I guess?  Same size/shape.  I flashed 0.8.6 to a bare ESP12e that I had laying around and have no issues with E1.31 or MQTT.  It didn't autodiscover in Home Assistant but I was able to manually add it and it works.

chickenzown, 2019-11-05:
I use Wemos D1 Mini,  I believe it is an issue with the use on pin 4 which is also part of the boot sequence, if you unplug the led strip, boot the device then hot plug the lights in it works fine,  the first LED on the strip lights up when stuck in boot loop. needs to be pulled high on boot, if low it fails to boot, which it appears to be by the onboard LED staying on.

Marlor, 2019-11-16:
I did some tests on a D1 Mini:

WLED_0.8.6_ESP8266.bin or WLED_0.8.6_ESP8266_APA102.bin flashed via basic "HTTP OTA updater" or a working WLED 0.8.5/6:
Bootloop after configurating WIFI

wled_0.8.6_ESP8266_lwip1.bin:
No Bootloop

Cleen WLED 0.8.6 Code compiled via VSCode and PlatformIO:
No Bootloop even though it is compiled with LWIP2

Maybe there is just an error in the release binaries?
The compiled .bin is with 680k even bigger than the 560k WLED_0.8.6_ESP8266.bin

marc-gist, 2019-11-16:
@Marlor interestings... I still can't get it to compile under VSCode :(  i still get errors with getUnixTIme, how did you overcome this.

Zaxxore, 2019-11-29:
I had the same issue (bootloop) with my D1 Mini v2 after a lot of try I compiled with setting lwip 1.4 high bandwidth. Flashed, with full erase set and now it works fine. Tks to Marc-gist. But, I didn't test all the feature.

closed 2020-04-05 by stale[bot]