wled/WLED #290: [Feature request] Make "on" property part of the segment

frenck, 2019-10-27:
Turning on/off is now considered a generic setting for the whole strip, IMHO it would be more useful if it became a property of a segment.

Right now, there is no way to turn off a segment individually. There is only a device level toggle for turning on/off the strip.

I think the device level decision of being on/off depends on the sum of the states from the segments.

The advantage of having this property as part of a segment is that we can start approaching segments as individually lights/strips.

Aircoookie, 2019-10-27:
This is very similar to your suggestion in #289 and also a good idea!

closed 2020-04-25 by Aircoookie

dmz006, 2024-02-29:
Was there any thought put into "I think the device level decision of being on/off depends on the sum of the states from the segments." - I use the relay control for the PSU for my LEDs and it would be convenient if the relay being on/off would be the sum of the segments on and not the switch for the entire strip.