wled/WLED #291: [Feature request] Allow for single time override transition time

frenck, 2019-10-27:
If would be useful if the transition time in the API could be overridden for just a single state change, ignoring the global transition property.

This would allow for turning light on instantly, e.g., alarm triggers, turn lights on NOW, full brightness and red (but there are more use-cases for this).

Aircoookie, 2019-10-29:
Added support for the `tt` key! Works same as `transition`, but doesn't change the default, so the value is only used for the one change.

frenck, 2019-10-29:
🎉 Really nice! 🚀 

Let's close up this issue, since well, it is done and you have enough open already 😉

closed 2019-10-29 by frenck

tonyn0, 2020-03-31:
Not sure if this post will be seen, but &tt **is** changing the default.

Aircoookie, 2020-04-01:
@tonyn0 agreed, confusing api design on my part.
&TT in HTTP API changes the default, while the `tt` property in the JSON API does not, but `transition` does instead. Will not change though due to compatibility.

tonyn0, 2020-04-01:
Ohhhh. Okay.

Suxsem, 2024-04-04:
@Aircoookie sorry does it still work?

I just built 0_15 and created a preset with tt instead of transition but when preset is called default transition time is changed. Thank you