wled/WLED #293: Nightlight target brightness can be 0, while brightness cannot be 0.

frenck, 2019-10-27:
According to spec, the target brightness of a nightlight is between 0-255. While the brightness property is between 1-255.

This is a mismatch.

Aircoookie, 2019-10-27:
0 is an accepted value for brightness, meaning off. You probably think that it is invalid because WLED returns not 0, but the last set brightness if on=false.

frenck, 2019-10-27:
No that is not it, it is documented that way, hence the two don’t add up. Either the code does not what it says in the docs, or the docs are not correct.

raoulteeuwen, 2019-10-27:
Changed doc. Hope it is o.k. for both @frenck and @Aircoookie .

Aircoookie, 2019-10-27:
@raoulteeuwen sure, thanks!

@frenck The section about brightness in the JSON API states:

| JSON key | Value range | Description
| --- | --- | --- |
bri | 0 to 255 | Brightness of the light. If _on_ is `false`, contains last brightness when light was on (aka brightness when _on_ is set to true). Setting _bri_ to 0 is supported but deprecated, it is recommended to use the range 1-255. The state response will never have the value `0` for _bri_.

This is exactly what the code does, and while it is correct that the JSON API will never respond with 0, the internal state of `bri` is in fact 0 when on=false and setting _bri_ to 0 is same as setting `"on":false`.
Were you referring to another page in the documentation?

frenck, 2019-10-27:
# `bri`:

> Setting bri to 0 is supported but deprecated

# `nl.tbri`:

> 0 to 255 Target brightness of nightlight feature

frenck, 2020-03-29:
Still an issue.

Aircoookie, 2022-10-21:
Closing as the wording of the documentation was changed before May 2021, indicating brightness 0 is a fully supported value :)

> Brightness of the light. If on is false, contains last brightness when light was on (aka brightness when on is set to true. Setting bri to 0 is supported but it is recommended to use the range 1-255 and use on: false to turn off. The state response will never have the value 0 for bri.

closed 2022-10-21 by Aircoookie