wled/WLED #294: Duplicate effects

frenck, 2019-10-27:
The `Chase` effect is on the list twice. For a normal user that is indistinguishable.

```json
{
    "effects": [
        "Solid",
        "Blink",
        "Breathe",
        "Wipe",
        "Wipe Random",
        "Random Colors",
        "Sweep",
        "Dynamic",
        "Colorloop",
        "Rainbow",
        "Scan",
        "Dual Scan",
        "Fade",
        "Chase",
        "Chase Rainbow",
        "Running",
        "Saw",
        "Twinkle",
        "Dissolve",
        "Dissolve Rnd",
        "Sparkle",
        "Dark Sparkle",
        "Sparkle+",
        "Strobe",
        "Strobe Rainbow",
        "Mega Strobe",
        "Blink Rainbow",
        "Android",
        "Chase",
        "Chase Random",
        "Chase Rainbow",
        "Chase Flash",
        "Chase Flash Rnd",
        "Rainbow Runner",
        "Colorful",
        "Traffic Light",
        "Sweep Random",
        "Running 2",
        "Red & Blue",
        "Stream",
        "Scanner",
        "Lighthouse",
        "Fireworks",
        "Rain",
        "Merry Christmas",
        "Fire Flicker",
        "Gradient",
        "Loading",
        "In Out",
        "In In",
        "Out Out",
        "Out In",
        "Circus",
        "Halloween",
        "Tri Chase",
        "Tri Wipe",
        "Tri Fade",
        "Lightning",
        "ICU",
        "Multi Comet",
        "Dual Scanner",
        "Stream 2",
        "Oscillate",
        "Pride 2015",
        "Juggle",
        "Palette",
        "Fire 2012",
        "Colorwaves",
        "BPM",
        "Fill Noise",
        "Noise 1",
        "Noise 2",
        "Noise 3",
        "Noise 4",
        "Colortwinkle",
        "Lake",
        "Meteor",
        "Smooth Meteor",
        "Railway",
        "Ripple",
        "Twinklefox"
    ]
}
```

Aircoookie, 2019-10-27:
Good catch! I will rename FX 13 to Theater Chase, I believe!

frenck, 2019-10-28:
`Chase Rainbow`, another one!

frenck, 2019-10-29:
Fixed in ddaaae46a6922cc79056756017be1eb1f431417b

closed 2019-10-29 by frenck