wled/WLED #295: [Feature Request]

Sander71, 2019-10-27:
First of all.. thanx for all the hard work.

I am using WLED with 12v pixels. It would be nice to use the  "current calculator" ( in LED setup) with these lights. Now they only use the 5v pixels. Maybe a volt selector ?

Than you again for thsi, i love it !

Aircoookie, 2019-10-27:
I'm very glad you like it!

Yes will add that! 12V pixels have gotten a lot more popular recently!

bagobones, 2019-11-30:
+1 for 12V as that is what I am running outdoors right now.. I noted that 24V strips exist as well so maybe this should be an expandable variable although I don't see anything out side of 5V/12V/24V being popular. 5V-24V is also the typical range for an off the shelf controller it seems.

blazoncek, 2022-07-12:
Automatic brightness limiter operates on current and supports any voltage.
For non-standard LEDs there is an option to enter custom current consumption (per single LED).

closed 2022-07-12 by blazoncek