wled/WLED #298: [Feature Request] Custom effects via scripting

bagobones, 2019-10-27:
ESPHome has lambda scripts that can create effects, due to various other features I like WLED much more however this is a neat feature.
https://esphome.io/components/light/index.html#addressable-lambda-effect

The idea being you can do a reusable effect you create and customize your self via scripting.

Aircoookie, 2019-10-27:
I've been thinking about easier to create custom effects. Lambda scripts sound like a very good starting point!

softhack007, 2023-01-30:
available as a usermod, currently only in the MoonModules fork:
https://github.com/MoonModules/WLED/tree/mdev/usermods/customeffects

https://github.com/atuline/WLED/wiki/WLED-Custom-effects

blazoncek, 2023-01-30:
Not sure it was a goo idea to close this issue as custom effects (via scripting) are not yet implemented in WLED.
MM or SR does not count, unless a PR is made from them.

softhack007, 2023-01-30:
yes you're right. 👍