wled/WLED #299: [Feature request] Support setting color temperature in mireds or kelvin

frenck, 2019-10-27:
I would love to see added support for setting a color temperature in mireds or kelvin.

This can be calculated (approximatively) for both RGB & RGBW strips.

Background information:

<http://www.tannerhelland.com/4435/convert-temperature-rgb-algorithm-code/>

The above will give an RGB, this can be calculated to RGBW like this:

<https://github.com/home-assistant/home-assistant/blob/master//homeassistant/util/color.py#L403>

Aircoookie, 2019-10-27:
Good idea! I will implement both a CT slider for the UI as well as a topic to set it from HA!

tonyn0, 2020-01-01:
Interested in this as well as color-correction! ;)

Aircoookie, 2021-08-31:
Small update on this, since it has been requested :)

A color temperature slider has been added in 0.11.0, setting a Kelvin value via API is already supported since build `2009260`, e.g. `"col":[[2700]]` for warm white.

Still _TODO_:
- Add this information to the API documentation
- Right now Kelvin is converted to RGB (and only then to RGBW, if enabled). A different conversion method might be possible for RGBW units, which makes more use of the relatively high-CRI extra white channel.

blazoncek, 2021-09-01:
If there will be changes to CWW logic, it would be good idea to add channel temperature (PWM CW & PWM RGBWW) in LED preferences page.

blazoncek, 2022-10-10:
Setting CCT in Kelvin implemented since 624993fc89266f4c206faeec535b542f52654519.
Behaviour explained in [KB](https://kno.wled.ge/interfaces/json-api/#cct-control).

closed 2022-10-10 by blazoncek