wled/WLED #300: Editable NTP-Server

riwich, 2019-10-27:
Since I run WLED (and most of  my other Smarthome devices) in a isolated network without internet access, the NTP Option as currently implemented (yes/no) is useless in my setup. 

Would be nice and helpful to have the possibility to edit the NTP-Server from the settings interface.

Aircoookie, 2019-10-27:
Sure, that makes sense! Will implement this when I rework the settings!

riwich, 2019-10-28:
Incredibly great software, amazingly productive author! Don't you need sleep, or are you even a robot? ;-)

Thanks for this fantastic piece of software.

Aircoookie, 2019-10-28:
Thank you so much for the encouragement!
I do in fact need sleep, otherwise we'd have segment support already! :)

riwich, 2019-10-29:
Ach was Aircoookie, Schlaf wird doch völlig überbewertet! ;-)

GrumpyMeow, 2020-12-19:
This feature is already implemented.

closed 2020-12-19 by Aircoookie