wled/WLED #301: [Feature request] Mosfet to disconnect leds when off

DotNetDann, 2019-10-28:
Long strings can actualy use quite a lot of power when they are off (most of the time)
Any chance that you can add a setting to toggle a pin when the LED's are off so a mosfet could be added to disable power to the string?

e.g. https://raw.githubusercontent.com/DotNetDann/ESP8266-MQTT-JSON-SK6812RGBW-HomeAssistant/master/Wiring%20Diagram.png

theshadowsvk, 2019-10-28:
There is a relay pin option, you can use it for this purpose. Btw it is easier to  cut power on 230V side(power supply side) which is much lower current than strip's current.

DotNetDann, 2019-10-28:
Thanks. I didnt see it in the settings. Didnt think to check the code for it. pin 12. Got it.

closed 2019-10-29 by Aircoookie