wled/WLED #302: [Feature Request] WRGB Yard Lighting

cabiv, 2019-10-28:
I would like to be able to use my WRGB as a yard light but do not need all the LEDs on at one time.   I would like an effect that turns on only the white LEDs leaving sets of LEDs completely off.

FX Speed could control the number of LEDs on in a row and FX Intensity could control the number of completely off LEDs between them.  

This could be done with just using brightness but I like the idea of not running a whole strip at a time.

I am envisioning running small groups of LEDs (1-4) with about 10 off LEDs between them.
W = White LED     * = LED off

**********WWW*********WWW**********WWW********

or 

*****W*****W*****W*****W*****W*****W*****

Aircoookie, 2019-10-28:
I like your idea, but this sounds like something where the upcoming Segments would also make a lot of sense. I'll think about it :)

cabiv, 2019-10-29:
Will look into Segments

Aircoookie, 2019-11-21:
I just added something very similar :) see https://github.com/Aircoookie/WLED/issues/343
Also will add another "Spots" effect that has the same function but will be easier to control with the sliders.

cabiv, 2019-11-22:
I will be looking into it.

Thank you

cabiv

cabiv, 2019-12-14:
Just downloaded 9.? Spots is at leat 90%+ of what I want.  Hope have time
over the holidays to compile my own - and tinker with a few mods,

Thanks for the great work.

cabiv

closed 2020-04-19 by stale[bot]