wled/WLED #304: Ledstrip freaks out on nodemcu when powered over usb

vwtune, 2019-10-28:
First of all, thank yo ufor this lovely project! I really enjoy it.

I encounter some problem dough;
In my test case i use ws2812B leds and a 5volt 20 amps power supply. The leds are connected on D4 pinout and the nodemcu is powered by the same powersupply as for the leds. I use pins vv and ground. WLED 0.8.5 is online and i can set the lights, but when i do around 70% of brightness in one the effect with alle the leds on, i can see the little blue led on de nodemcu is almost off (dim) and WLED is slow or just crashes.

Is there a special connection of the nodemcu to your leds and powersupply that i'm missing?

Also i tried to powered up the nodemcu by usb (tried a couple) but then wled just freaks out, all the leds will start to flicker and i can not control wled. Why is this?

I tried wemos D1 mini's, but wled will start up, but than it will freaks out also ( on 5 volt and powered by usb)

Can you give me some hands up here please?

raoulteeuwen, 2019-10-28:
- do you have a ground running directly from the nodemcu to the LED ground?
- how many LEDs did you connect?
- what is 'vv'?
- sorry for maybe asking the obvious, but you do send in the data on the right side of the LED-strip?

vwtune, 2019-10-28:
> * do you have a ground running directly from the nodemcu to the LED ground?
> * how many LEDs did you connect?
> * what is 'vv'?
> * sorry for maybe asking the obvious, but you do send in the data on the right side of the LED-strip?

Powersupply
+ -->nodemcu (5vin/vv) and ledstrip
- --> nodemcu (Ground) and min to ledstrip
vv is 5v in on the nodemcu

I use 5 meters 60 leds, so 300 leds are connected.
The green (data) is set on the right side

raoulteeuwen, 2019-10-28:
Just checking (and if you have wired it differently, could you do it like this as a test): is it like in https://www.reddit.com/r/arduino/comments/bzv5tx/does_this_wiring_look_correct_for_a_nodemcu/ , with the exception that the green data wire is running from D4 on the NodeMCU to the LED data cable? Can you share a picture?

vwtune, 2019-10-28:
Yes, it is connected just like the picture shown in your link!
The problem withe the wemos d1 mini and the overall problem with powered over the usb connection, i strange, but why the nodemcu is struggling ( as i can see when the blue light of the esp chip will fade out) when the brightness increase. the powersupply have enough spare amps for the strip so i don't get it

raoulteeuwen, 2019-10-28:
Do you have a strip with fewer LEDs to test with?

vwtune, 2019-10-28:
> Do you have a strip with fewer LEDs to test with?

Yes, and it then i have no problems when test it with a 24 leds ring. Also the leds of the led ring is much brighter when brightness is set on full (i can not look straight in the led).

Again, my powersupply is a 5volt 20 amps. 
the 60 led p/m draws 6 watts

raoulteeuwen, 2019-10-29:
Based on what you shared, i would say either the strip is drawing to much current (maybe more than specified) and/or your power supplies deliver less than spec'd. I would not know how to measure nor what equipment that would take. I always read that when you increase the nr of LED's and length, the chance increases that you will need to inject power at more points in the strip. I'm pretty sure it's not a WLED thing.

vwtune, 2019-10-29:
> 
> 
> Based on what you shared, i would say either the strip is drawing to much power (maybe more than specified) and/or your power supplies deliver less than spec'd. I would not know how to measure nor what equipment that would take. I always read that when you increase the nr of LED's and length, the chance increases that you will need to inject power at more points in the strip (more with 5V strips, less with 12V strips). I'm pretty sure it's not a WLED thing.

Okay, but why de nodemcu is affected when i pull up the brightness? You can see the blue led of the esp8266 on the nodemcu fade out.

also on a wemos d1 mini, with the same power supply and a 24 led ring, i only get flickering of all the leds, i can't controll the WLED at all (it is online though)

raoulteeuwen, 2019-10-29:
I hope someone else will reply to the info above ... i'm not an electrical engineer, but that's why i wanted to make sure you have the ground running correctly. Not sure whether all connections are sound? I have a NodeMCU with 120 LEDs running fine, and based on this forum i would say there are hundreds of people using NodeMCU and Wemos D1 with WLED, so the only thing i can think of is something with wiring or power (either supply or usage). But maybe someone else has another idea of what could be wrong...

mjferro, 2019-10-29:
If you have a multimeter you can check to see if your getting a significant voltage drop from your power supply.

As for the flickering, if you power the node mcu with usb and the strips with your separate power supply,  make sure the grounds are connected between the two different power sources.

vwtune, 2019-10-29:
> 
> 
> If you have a multimeter you can check to see if your getting a significant voltage drop from your power supply.
> 
> As for the flickering, if you power the node mcu with usb and the strips with your separate power supply, make sure the grounds are connected between the two different power sources.

I measured the power supply when connected the nodemcu and 300 leds running, power supply stays steady on 5 volt

So, when connected a wemos (or nodemcu) with USB, you also have to connect ground to it from your power supply?

mjferro, 2019-10-29:
When using multiple power sources, you still want to have everything on the same common ground. Otherwise there may be voltage difference relative to the other supply, or other weird stuff.

Try powering the nodemcu via usb, and the strip with the 5v power supply. Then connect the grounds together (strip gnd to mcu gnd).

eoncire, 2019-10-29:
> I measured the power supply when connected the nodemcu and 300 leds running, power supply stays steady on 5 volt
>
> **### So, when connected a wemos (or nodemcu) with USB, you also have to connect [...]

Yes.

thedannymullen, 2019-11-27:
@vwtune have you resolved your issue?

closed 2019-11-28 by vwtune

vwtune, 2019-11-28:
I powered the nodemcu apart (by usb) and the leds by powersupply, only the ground is connected together from the powersupply to the leds and nodemcu, now i don't have the problems anymore.