wled/WLED #305: Send device MAC address in zeroconf/mdns announcement.

frenck, 2019-10-28:
It would be great if the MAC address of the WLED device was send out via the zeroconf publish announcement.

While the current situation is great to find instances of WLED on the network, there is no way to figure out if the device you've found on the network isn't already connected to your application. (Considering the MAC address is the only unique factor here).

The only workaround right now is to connect to the discovered device to get the MAC address via the API. However, if it would be in the discovery information already, that would not be needed.

frenck, 2019-10-28:
Looking at the codebase, it is probably a single line right below this one:

<https://github.com/Aircoookie/WLED/blob/f4a2ffc5d29f77675ed012fdb17c40a6df5e622e/wled00/wled05_init.ino#L219>

Something like this? 🤷‍♂ 

```c
MDNS.addServiceTxt("wled", "udp", "mac", escapedMac.c_str());
```

Aircoookie, 2019-10-29:
Added! I always thought that IP was enough as an unique identifier (with most routers anywats), but of course that could change with DHCP so this is a welcome addition!
(by the way, your one-liner indeed works!)

frenck, 2019-10-29:
Thanks! Well, I run into this while testing the Home Assistant integration.
The device will be in there for a long time (so indeed the IP address might change). Secondly, the hostname is changeable by the user in the WLED interface, so, that leaves the MAC address as the only really immutable identifier.

I want to make sure a device is not discovered over and over again, while already configured in Home Assistant (and thus generating a notification). Also, I want to prevent a user adding a device that exists already (e.g., by adding it by IP address manually).

Nevertheless, thanks for adding this! 🎉

closed 2019-10-29 by frenck