wled/WLED #306: power cycle makes WLED "forget" previous (wifi/mqtt) configuration

prankousky, 2019-10-29:
Hi everybody,

I just flashed the most recent pre-build bin to a `nodeMCU`. It will open an AP and let me enter my wifi (and mqtt) credentials. When I then restart, it will connect to my network (I can ping it and `wled/#` will receive mqtt payloads).

When I then unplug it and attach my `WS2812B` 8 led stick (GND => GND, 5V => VCC, Pin D4 => IN) and start it up, it will not connect to my network any longer! Instead, it will open an AP again.

When I connect to that AP, I will be able to control the ws2812; I can change colors, start/change effects, etc. etc.; but I need the device to be in my network, not to always connect to its AP. Anybody know why this might be?

What I did and am using

* `sudo esptool.py --port <port> erase_flash`
* `sudo esptool.py --port write_flash -fs 1MB -fm dout 0x0 WLED_0.8.6_ESP9266.bin`
* device `nodeMCU v3`
* power supply / extender `NodeMcu Ver 1.0`

The nodeMCU is inserted in the power supply / extender; this will allow me to power the nodeMCU with a power supply that is powerful enough to also power larger LEDs; I am successfully using this very power supply with multiple `ESPHome` nodeMCUs connected to much larger LED strips, so it should be sufficient for this as well (also, WLED *does* work when I control it via its AP, so it looks to me like it does get sufficient power).

Did I make an error erasing and/or writing the flash? The command I used is taken from the `Tasmota` wiki; I have previously flashed countless devices (ESP8266, nodeMCU, all sorts of `Sonoff` devices, etc. etc.) with this very command and it worked fine. But I don't understand why else the device would "forget" the  entire configuration after I power-cycle it just once.

Thank you in advance for any ideas on how to fix this :)

prankousky, 2019-10-30:
Thank you! Yes, the device will stay in the network until I turn it off again.

I am not sure if I understand correctly... Here is what I did

* connect to AP
* enter my WiFi credentials
* save & reboot
* connect to device via WiFi
* change some other settings
* save & reboot

I cannot ping the device and it opens the AP again. **However**, once I accesss the device via WiFi, the correct credentials (or at least the network) are displayed. So they did save (or does the device not reboot upon me entering the credentials in AP mode?).

I should probably also mention that the device is forbidden to access the internet. This **shouldn't** matter, but perhaps it does?

raoulteeuwen, 2019-10-30:
I get the impression @Aircoookie is already on/knows about the problems with saving config, so for now i would wait for his findings. Maybe watch https://github.com/Aircoookie/WLED/issues/271 ...

Aircoookie, 2019-10-31:
@prankousky thank you for the report! Yes, very similar issues were also reported, #271, #297, and #309. You could try two things:
- Try flashing the bin with less parameters. This should also work and may fix the flash:
 First: `sudo esptool.py --port <port> erase_flash`
 2nd: `sudo esptool.py --port write_flash  WLED_0.8.6_ESP8266.bin`
- Try the binary with the `lwip1` suffix.

I hope we'll be able to solve this issue :)

prankousky, 2019-10-31:
Thank you @Aircoookie 

As far as I can tell, it works now. Reboots seem to survive reboots. Awesome :+1:

closed 2019-10-31 by prankousky

paulfecht, 2019-10-31:
Settings saved and persist after reboot here.  My Binary was:  lwip1

scargill, 2019-11-01:
I tried BOTH files this morning (1/NOV/2019) - set the unit to rainbow, set edit preset, pressed 1, unset edit, save.... lights were on rainbow (I could see)....  reboot - back to solid orange in both cases... am I missing something? If I give it a preset to use, then after power up it WILL retain rainbow, but if I leave that setting at zero and tick "Set current preset cycle setting as boot default" and then reboot - when I  come back it is back to solid orange and that tick box is unticked.

Also, WLED_0.8.6_ESP8266.bin - Alexa, change name to "lighting" - save, reboot - still says lighting, but tell Alexa to find devices... nothing. Tell Alexa to "turn off lighting" - nothing...

Duckshousemacservice, 2019-11-05:
i tried every version (085 and 086) on 2 wemos d1 mini and they forget everything after reboot.  i have one running on 0.8.5 (build 1909121) and thats ok -- but i cant find that build anymore  must have deleted the bin.  otherwise super great software. i love it.