wled/WLED #307: Question: About ap opens = never

relax81, 2019-10-29:
Hello, 
I'm fairly new with WLED. Do i understand it correctly that when i set it "AP opens" to "never" that it then will keep reconnecting to the previous set Wifi Network until it's successful? No matter what happens?

I've used previously McLightning for some lightings that are installed at my parents and the AP mode was the biggest issue. Whenever there was a Router update or internet issue it would start to go into AP Mode and they were lost. They've no idea what an AP is or how to deal with it.

So i need a setup that will keep the same wifi credentials forever once it's setup once. Maybe with a hidden button to trigger AP Mode, but that wouldn't be necessary. 

Thanks

Aircoookie, 2019-10-29:
Hi and welcome!
Yes, never open AP does what it says on the tin! It will never open the access point, except for when the (optional) button is pressed for at least 6 seconds.
However in WLED 0.8.6, unlike in many other projects and previous versions of WLED, the module will even attempt reconnecting to the saved network if the AP is open.

Hope I was able to help!

relax81, 2019-10-31:
Thank you, that was very helpful. Then I'll switch all of my projects to WLED :)

Thanks for the hard work behind this - it's a beautiful project.