wled/WLED #308: Full UI password protection

jj-uk, 2019-10-29:
v0.8.6
I know it's a hassle, I read the notes in the docs about security, but can we please get a password to access the UI ?

Every device on my network has a password, except this. The fact that it cannot be re-flashed without a password is great, but i'd like the whole UI password protected.

My biggest threat is from within my network - teenage kids! They bring round their friends with their laptops to do homework etc..  If one of them scans for IP addresses, and then messes with my settings, i'm gonna be really mad.

I can't ban all my kids' friends, so my best defence is to ensure all my devices are secure.

Aircoookie, 2019-10-29:
I will consider this! AFAIK the ESPAsyncWebserver already has an option for password protecting pages. It would be too much of a hassle to always enter the PW for me personally, but having the option to do so (like in your case, where you can't be sure there are no unauthorized users on the network) can't hurt.

jj-uk, 2019-10-30:
Thanks. My browser will take care of entering the password each time.

pepeEL, 2019-12-28:
Good option and should be add

rant989, 2020-03-15:
I live at home with partners
Our router is unable to set up a VLAN and block access so that anyone in the home or guests receiving WIFI access can control the LED lights in the room

mew1033, 2020-11-06:
@Aircoookie I know previously you mentioned API keys, which might be a nice way to allow programmatic access (Home Assistant) while still locking down the UI.
Maybe an IP whitelist could be a good interim solution? Only allow access from specific IP addresses, but don't worry about adding a password to the UI.

pepsonEL, 2020-11-10:
Is any chance to add password protection UI ?

blazoncek, 2022-07-25:
#2610 and #2656 will address protecting Settings page.