wled/WLED #309: wifi settings 0.8.6

lpacchin, 2019-10-29:
NodeMcu esp8266 12E 4M

Flash 0.8.6, I use already compiled bin

when I put my wifi settings, then there is no way to start, checked with serial monitor, nothing happen, stuck, changed to another board same thing

I can only use in ap mode and is working fine

I flashed 0.8.5 everything is working fine, wifi settings, dhcp working, everything working
Thank you very much

Luca

paulfecht, 2019-10-30:
I seem to be having a similar issue.  I flash it and then get stuck on the "WiFi Settings saved.  Reconnecting now . . .  " screen.  WHen I go in under my networks and connect to it directly I seem to be able to use some of the functionality....  I'm really only putting in my SSID and my SSID password..... once in a while the pass: wled1234 from the Wiki.... in short 0.8.6 seems to work slightly differently.  Any help would be appreciated.  thanks.  p.s. 0.8.5 works just fine for me.  I'm flashing it with ESPHome-Flasher-1.1.0-windows-x64.

raoulteeuwen, 2019-10-30:
> I tried to flash this version 0.86 last night but I get stuck at wifi settings.. say reconnecting now.. but actually doesn't reconnect.. so no IP .. and not working ! Any update or suggestion ?

For now you might want to flash 0.8.5 and enjoy that ... dev knows about the problem...

Aircoookie, 2019-10-30:
@lpacchin Thank you for the report! Did you compile from source or bin? Does saving other settings (like changing LED count and saving LED settings) work?
@paulfecht For me the browser won't refresh (aka will stay on the "Reconnecting now" screen indefinitely), but the ESP will actually connect and be reachable from your network. But i believe you already checked that, so it seems like your ESP can't connect to the netwok. If you power-cycle the ESP, do the changes you made to WiFi settings also get lost? You could also try if the binary with the `lwip1` suffix makes a difference!

Thank you so much for your help! WIthout your feedback it would be near impossible to crack down this issue (since I can't reproduce it myself)

paulfecht, 2019-10-31:
Thanks for the response.  I tried flashing with the lwip1 binary.  Here's how my experience evolved(iPhone 7): 1.) I clicked on the WLED app icon  I get the  " Found no lights yet..." but my lights are plugged in and glowing amber color.  30 of them are lit up by the way.  It seems when I switch between the WLED screen and the " + " button in the upper right hand corner I see the flash of a light wanting to be shown but it won't show.  2. ) since this doesn't seem to work I head to the iOS  Settings > Wi-Fi > Choose a Network and "WLED-AP" is listed.  I touch it.   After a pause I get a screen which at the top reads 4.3.2.1  WLED-AP  Log In with the Smiley Face and then at the bottom I get the two buttons "Wifi Settings"     " To the Controls"  ..... so I select WiFi settings.    I enter my SSID and the SSID password.  I scroll to the bottom and notice that the field  AP opens: No connection after boot is the default.  I do nothing else on this page but enter SSID and password and click the button " Save & Connect"  .... Then I get the  "WiFi Settings saved reconnecting now"   after about 15 seconds I'm brought back to the Wi-Fi screen for iOS.  I assume I'm connected so I head back out to click on the WLED app icon.     IT WORKS!    For me the solution was the lwip1 binary.  At least I believe it was.  I'm using the ESP Node MCU 4mb.  Thanks for your help.  I'm back up and running.

Aircoookie, 2019-10-31:
@paulfecht Thank you for the detailed description. Awesome that you got it to run again!
Seems like sticking with lwip1 is a solution to some of the issues.
One thing I'm interested in is whether the wifi credentials saved, because some users were reporting that wifi settings weren't actually saved (#297 and #306). It would be awesome if you could disconnect your ESP from power once and see if it, after rebooting, is still accessible via the app :)

lpacchin, 2019-10-31:
@Aircoookie Thank you, I've done many tests, with lwip1 everything ok, with regular WLED_0.8.6_ESP8266.bin does not work after putting wifi settings in ap mode, then I compiled with arduino and put wifi settings in WLED00 after flashing it is running fine!
Flashed another esp8266 4M and also this one works perfectly so problem solved, I'am sure WLED_0.8.6_ESP8266.bin already compiled have some problems
Now I can go further with API, MQTT
Thank you very much!

closed 2019-10-31 by lpacchin