wled/WLED #311: Sunrise effect

Aircoookie, 2019-10-31:
Planning to add https://www.youtube.com/watch?v=2j3UyrBjB4g

eliobou, 2019-11-01:
I'd love to have this feature, do you think it's possible to do it without RGBW but classical RGB ?

Aircoookie, 2019-11-02:
Yes, that is exactly the problem I need to solve. Of course this effect is made for RGBW, so it will look better and brighter on it, but for it to be a part of WLED I need to achieve proper RGB compatiblility!

Mariu86, 2019-11-04:
Aircoookie thanks for all the good work, and if you working on sunrise efect i think a sunset efect will be awesome for a child like my 3 year daughter. The sunset i think will look more natural then Nightlight function and if you can put the time delay for 1 hour will be enough for my daughter to go to sleep , thank you very much Aircoookie !

TimmyUll, 2020-01-08:
I love this! Hopefully this will still work at full capacity with my rgbw! :)

smoke007, 2020-02-14:
I have parts on order to install sunrise simulator RGBW LED strips in 3 bedrooms. Would love to be able to use WLED for this project when this feature gets added. Thank you in advance!

Yusef1985, 2020-02-22:
I would love to see this effect in WLED, i have been struggling with https://www.youtube.com/watch?v=2j3UyrBjB4g for 4 weeks do get this feature to work and still no luck :(

DieNummerEins, 2020-02-26:
Watching!

Cpainy, 2020-04-02:
I would love this as well!! Such a cool effect for my WWA strip

thedannymullen, 2020-04-09:
> I would love to see this effect in WLED, i have been struggling with https://www.youtube.com/watch?v=2j3UyrBjB4g for 4 weeks do get this feature to work and still no luck :(

What issues are you having?  I played hell getting the software in the youtube video to work.   In the end I did though.  Did you get it?  
Still would love it in wled.

Yusef1985, 2020-04-12:
> What issues are you having? I played hell getting the software in the youtube video to work. In the end I did though. Did you get it?
> Still would love it in wled.

I never got MQTT to work, so I could not get it to start at a specific time :( I hate giving up but I had I this time.

Yusef1985, 2020-04-12:
> Planning to add https://www.youtube.com/watch?v=2j3UyrBjB4g

It could be fun to hear if there is any progress with this project? 
:)

smoke007, 2020-05-04:
Macro `CL=hff0000&C2=hFFCE2E&A=5&NL=10&NT=255&NF=2` looks pretty good on RBG until The Hook Up's sunrise effect can get ported over. (From) [https://github.com/Aircoookie/WLED/issues/470](url) (`CL=h000000&W=255&A=5&NL=10&NT=128&NF=1` for RGBW)

thedannymullen, 2020-05-07:
@smoke007 this work on rgbw?

smoke007, 2020-05-09:
@thedannymullen I updated my post to make more sense and I see the link to my source wasn't working correctly. The first macro is for RGB and the second macro is for RGBW light.  Unfortunately, I still haven't gotten my RGBW lights in yet, so I can't test that macro.  

If you want to learn what each of these macro parts are doing, you can look them up here.  https://github.com/Aircoookie/WLED/wiki/HTTP-request-API

The easy way to test your macros is by using them in a web browser.  Get the IP of your WLED and swap out the IP in the following URLs:

RGB lights:
192.168.4.1/win&CL=hff0000&C2=hFFCE2E&A=5&NL=10&NT=255&NF=2

RGBW lights:
192.168.4.1/win&CL=h000000&W=255&A=5&NL=10&NT=128&NF=1

For testing, I set my &NL = to 1 instead of 10 so it would do the animation over 1 minute instead of 10 minutes.