wled/WLED #313: 0.8.6 Issues on NodeMCU

scargill, 2019-11-01:
Hi
0.8.6 newby here. If I set a sequence like RAINBOE - then in settings tick say I want the current sequence on powerup - then hit SAVE, I'd expec the rainbow after a power cycle. It isn't happening,

Also Alexa, using the default name, Alexa turne WLED off, but would not turn it O. I changed the Alexa name to "strip" and SAVED, then told Alexa to "find devices". Nothing happened and when I said "alexa, turn off STRIP", Alexa said it could not find a device called STRIP. Is it me? I've used Alexa with real devices, Node-Red virtual devices and Sonoffs MANY times. ???

raoulteeuwen, 2019-11-01:
Hi @scargill There are some problems with saving settings. You might want to check https://github.com/Aircoookie/WLED/issues/306#issuecomment-548251394 .

scargill, 2019-11-01:
I've put my comments in 306 but that seems to be closed, so - here they are...  I tried BOTH files this morning (1/NOV/2019) - set the unit to rainbow, set edit preset, pressed 1, unset edit, save.... lights were on rainbow (I could see)....  reboot - back to solid orange in both cases... am I missing something? If I give it a preset to use, then after power up it WILL retain rainbow, but if I leave that setting at zero and tick "Set current preset cycle setting as boot default" and then reboot - when I  come back it is back to solid orange and that tick box is unticked.

Also, WLED_0.8.6_ESP8266.bin - Alexa, change name to "lighting" - save, reboot - still says lighting, but tell Alexa to find devices... nothing. Tell Alexa to "turn off lighting" - nothing...

raoulteeuwen, 2019-11-01:
Fyi: i'm still on 0.8.5 and also can't use 'use current setting on boot' > on boot, it is back on another setting. So it was already a problem in 0.8.5. If you follow this github, we will see what the findings of @Aircoookie are relating storing/saving settings like this.

lpacchin, 2019-11-01:
Hello I'm using 0.8.6 and confirm WIFI, MQTT, boot FX, ecc settings are all saved after reboot
Also the wifi on 0.8.6 do not loose connection
I suggest to not use the bin already compiled WLED_0.8.6_ESP8266.bin but try to compile yourself and flash your nodemcu with arduino

lpacchin, 2019-11-01:
@scargill 
Tried also Alexa, changed name, find devices, found new with the new name, told to switch on and off and is working fine

scargill, 2019-11-03:
I saw something about an update but I'm on 0.8.6 which is days old and I dont see anythin later. Do we know when an update will be out.. Also am I the only onw who doesnt fully understand the OTA procedure? Is it automatic unless blocked? I see in settings something about changing passwords but the only passwod i've entered is for the access point so it isnt obvious if I'm supposed to trigger an OTA. somehow.

brentbrooks70, 2019-11-03:
There are 2 types of OTA updates available on WLED
1) Manual OTA, which allows you to upload a new bin file to http::/DeviceIP/update, this should always be locked via a password on the settings-> Security & Updates page to prevent anyone else from uploading erroneous firmware to your MCU. 
2) Arduino OTA is also available and shows up in the Arduino IDE just like a Com port does, from here you can select the board and push new firmware directly from the Arduino IDE, this only applies if you are compiling yourself and not using the bin files provided. 

The OTA lock prevents either of these methods from working until you are ready to change firmware, at that point you can unlock it by way of the password, flash your update via either OTA method and then re-lock the device so it is secure until you are ready to update gain. 

There is no automatic update process, you will have to at minimum download new code or bin file, and compile and upload or just upload bin file, assuming the OTA lock is not enabled.

Hope that helps.

closed 2020-04-05 by stale[bot]