wled/WLED #314: New light control interface - Home Assistant API

Nornode, 2019-11-01:
I'm thinking of testing this out as I'd like better support for LED strips specifically. I know it would be very much easier to integrate into Homeassistant with its own HomeAssistant API  much like ESPhome has and adoption rate with HA users would rise!

It is currently possible but the configuration with MQTT isn't for everyone and adoption is much slower with a lot of troubleshooting involved if it's not working correctly.

Please consider supporting native HomeAssistant API.

TheZoker, 2019-11-07:
I think this issue can be closed :wink: 
https://github.com/home-assistant/home-assistant/pull/28542

Nornode, 2019-11-07:
Woohaa! Awesomeness!! 😍👍

closed 2019-11-26 by Aircoookie