wled/WLED #315: Alexa RGBW Color Support

Def3nder, 2019-11-01:
Hi Aircooockie,

I've attached an RGBW LED strip to WLED on an H801 Wifi RGBWW controller that works just fine.

However, when controlling the color with Alexa and the Hue emulation, the white channel isn't used at all.
So, if I say "Alexa, switch the light to white", I do get a (very cold) white purely from the RGB leds and the (warm) white led stays off.

The EspalexaDevice works with RGB only - would it be possible to introduce RGBW here as well ?
(like in the rest of the WLED code)

Aircoookie, 2019-11-02:
Hi!
The white channel should automatically be calculated from RGB values (when directly setting from  Alexa)
If you set your Hue lights via Alexa and have WLED sync to them, then you are correct, for hue sync the white channel is inactive. I should change that :)

Def3nder, 2019-11-04:
Hi Aircooockie,

I do not have any Hue lights - but I do have a couple of ESP's that publish to Alexa via Hue-Emulation.

Same I did with WLED and for me, the white channel had not been used at all when pushing a white color to WLED. It was allways a combination of the RGB channels.

I've tweaked the code in ESPalexa-device.cpp to use the white channel at 100% and adding either pure RGB-white (255,255,255) to it (when requesting cold white) or adding RGB-orange to it when requesting warm white. Result is that the white LEDs are on at 100% allways and only for the warmer / colder colortones the RGB LEDs will be added as well.

I do have the code on my laptop - I will publish it on Github Gist when I have access to my Laptop again...

Greetings from Berlin :-)

Def3nder, 2019-11-06:
Hi Aircoookie,

I've added the code to a new branch in a fork - I would appreciate iof you could have a look ?
https://github.com/Def3nder/WLED/commit/53e3fc29bfb64eb212862a7114119b42c815f5ea

I've changed ESPAlexa to work with 4 colors rather than RGB only.
The reason behind this is, because converting from a colortemperature from RGB to RGBW will never use 100% in the white channel. 

If you like I can do a pull request ...

closed 2020-01-07 by Aircoookie