wled/WLED #316: [Feature Request] Battery status?

OliPassey, 2019-11-01:
I'm running a small setup of a 16bit neopixel ring & nodemcu off 2x 18650 batteries, i see that the NodeMCU should support returning a battery status, would it be possible to surface this somewhere in wled - potentially sending a notification when battery is low, or shutting off before a battery is completely drained?

Aircoookie, 2019-11-02:
This is a good idea. The NodeMCU can read out the battery voltage at the A0 pin.
Making that available via the API and setting a voltage cutoff for turning off LEDs shouldn't be too difficult.

Riza-Aslan, 2020-07-22:
Any news on this? Would also like to build a battery powered LED controller and the battery status would be really great to have.

@OliPassey Do you have pictures of your setup and would tell me which components you have used for it?

OliPassey, 2020-07-22:
Would be great to get this exposed via MQTT so we can alert on low battery.
Pics of my various creations: https://photos.app.goo.gl/kQUD2KcpC8NwBnnC6

I'm just using NodeMCU v2's, WS2812b 16bit Neopixel Rings, Sanwa Arcade buttons & 3d printing the enclosures.

OliPassey, 2020-07-22:
These are also very useful, i use the dual battery holder ones but they make them up to 4x 18650s.
https://www.ebay.co.uk/itm/ESP32-For-Wemos-For-Raspberry-Pi-18650-Battery-Charge-Shield-Board-V3-Micro-LS/174358018669?hash=item28988c3a6d:g:v-gAAOSwictd3kO7

supersonical, 2020-12-23:
This would fit my wled project perfectly, wish that this enhancement would be bumped up 😀

stefankoehne, 2021-04-08:
Is there anything new here? Battery status would be awesome :)
Multiple board support could be done with different voltage dividers between VCC and A0, e.g. 180 kOhm for full 5V range (D1 mini). The idea from @OliPassey is also great to monitor battery voltage via MQTT.

Aircoookie, 2021-04-08:
Nothing new from me, but this feature would be great as a usermod. Maybe I have time at some point to implement that, but if someone wants to give it a try, it shouldn't be too difficult :)

Some ideas:
- Read the A0 pin and convert the reading into battery voltage depending on the voltage divider used
- Display that voltage in the info section (e.g. the "Temperature" usermod already does that) and/or map the voltage to an approx. 0-100% depending on battery type.
- Optionally, one could also set a voltagethreshold for turning off (or set to red blinking) the LEDs (they would still be powered, so still need to disconnect the battery) to reduce the likelyhood of an over discharge.

rocket-hub, 2021-04-13:
Hi, I enter the discussion ...
I have made some battery systems with WLED, and I have the same need to monitor the battery, it would be fantastic to be able to view it in the node list, where each node also gives its level.
thank you

ybarigou, 2021-05-02:
+1, would definitely be a great addition to wled, I have batter-based wled projects and would love to be able to monitor status of battery.
Thanks!

itCarl, 2021-07-29:
i would really need this right now.

nickizgr8, 2021-08-19:
Has there been any developments on this? :) Having the capability to read battery voltage and/or level % via the WLED app would be OK. being able to view this information in Home Assistant would be amazing! Additionally being able to have the leds play a saved favorite (blinking red or any favorite you choose) when the battery gets below a specifiable level would be cool too. Someone could make an automation to do similar in Home Assistant based on the battery level but being able to set this up via the app would be great too. Can you run ESPHome and WLED on the same ESP8266 or -32 somehow? If you were to copy the bin file into ESPhome and upload it from ESPhome adding a temeperature read would be easy enough but would this work?

itCarl, 2021-08-19:
yes, i wrote a basic usermod to read the battery level/voltage #2127
since i dont really use any Smart home software like "HomeAssistant" i can't really develop and or test any feature like that.

OliPassey, 2021-08-19:
Amazing!
Will give this a try tonight, thanks @itCarl  !!

blazoncek, 2021-08-19:
@itCarl it is rather simple, just add the following in a loop (and broadcast only every minute or so):
```
if (WLED_MQTT_CONNECTED) {
  char subuf[64];
  strcpy(subuf, mqttDeviceTopic);
  strcat_P(subuf, PSTR("/voltage")); // or battery_level or whatever
  mqtt->publish(subuf, 0, false, String(voltage).c_str());
}
```

blazoncek, 2021-10-03:
Implemented as a usermod included with master since 0.13-b3. Closing.
Feel free to reopen if needed.

closed 2021-10-03 by blazoncek

itCarl, 2022-04-03:
Unfortunately, due to work I currently have little time to implement your suggestions and ideas. Therefore someone else can try to implement the ideas

VASILBEAM, 2022-09-26:
hello, is it possible to display battery information on the display in the user display mode. Can someone help me with this?

itCarl, 2022-12-29:
> > * Optionally, one could also set a voltagethreshold for turning off (or set to red blinking) the LEDs (they would still be powered, so still need to disconnect the battery) to reduce the likelyhood of an over discharge.
> 
> any update on this idea? Would be super useful to add it :)

The Battery Usermod got Updated recently.
In case you can't find it in the usermod folder i renamed it to just "**Battery**" before it was called "battery_status_basic".

⚙️ _New features:_
- 🕙 Shows the next time the statistics in the info modal receive an update
- 🚫 **Auto-Off** => Turn off leds at a configurable Threshold
- 🔧 **Calibration** value => a configurable value for the correction of voltage measurement errors
- 🚨 **Low power Indication** => Indicates low power with a given preset at a configurable Threshold
- (Generall optimizations)
- (Info for developers 👨‍💻 👩‍💻 ) The Battery Class is now open and exposes all necessary member variables through Getters, you can also set some values via Setters (As example this way you could simulate having a battery connected). You could even write a Usermod that depends on knowing the battery voltage. So now instead of implementing a voltage read feature you can just grap the values from the Battery Usermod.

for more information have a look at the [Battery Usermod](https://github.com/Aircoookie/WLED/tree/main/usermods/Battery)

PS: 
The next thing on the list would be to calculate the "remaing battery time". Unfortunately, I haven 't managed to add this feature to the current update, due to lack of brain mass. I already wrote a little text about this problem in my [PR](https://github.com/Aircoookie/WLED/pull/2975) below the Other section. If you have a better Idea please let me know. Thanks ❤️

philimi, 2022-12-29:
Nice update! Looking forward for the "remaining  battery time". I use this mod a a a key feature in my project so this will be a great feature. I had the same idea with average of the current. As "cutoff" voltage I would assume 6.0V,personally I wouldn't go lower then this and most BMS also will cut off.

madjetey, 2023-05-04:
Is battery still exposed to MQTT?  I've seen the [merge](https://github.com/Aircoookie/WLED/pull/2168) but it's not showing up in HA
[image]
[image]

blazoncek, 2023-05-04:
As I see it, it does not advertise itself. Add it to HA manually.