wled/WLED #317: All work well, but Alexa replies "device not responding" to every command

raptusis, 2019-11-01:
As the title suggests, everything works well and the controls are turned on and the colors are changed, but Alexa replies "the device does not respond ... etc..." to every command.

I have echo dot 3° gen.

I tried everything, what could I do?

Thanks in advance and compliments for all this project

raoulteeuwen, 2019-11-01:
Hi @raptusis . Maybe you wanna watch https://github.com/Aircoookie/WLED/issues/274#issuecomment-548400806 ...

scargill, 2019-11-01:
I didnt't see any answers in 274.. I have the same issue- Alexa not working. or perspective I use Tasmota on various boards - and Espurna, both (as of recently) ork with Alexa.

raptusis, 2019-11-02:
Thank you SCARGILL ! 
I didn’t know that Tasmota have Alexa integration.
Thanks ;-)

Aircoookie, 2019-11-02:
See https://github.com/Aircoookie/Espalexa/pull/88! Someone found a fix, I will release a new espalexa version which will fix this issue as well

@scargill so discovery is not working for you at all?
I'll consider the multi wifi support, but still i believe that for portable installations (cosplay, vehicles...) AP mode is better suited. For fixed installations, just connecting to the strongest AP at that location will be enough. Still it might have some uses!

Aircoookie, 2019-11-02:
Closing as it is a duplicate of #274 :)

closed 2019-11-02 by Aircoookie

scargill, 2019-11-03:
~Thsnkd -will check 274 thanks on Alexa. Comment only coming up: From experience, I'm not by any means the only one who uses multi-access point names in fixed installations -My own ESP software and Tasmota  offer 2 access point client names, Espurna goes further. If I may, I really think it would be nice but understand it is down to time.  I'm sittin here surrounded by rainbow lights. Lovely, well done.