wled/WLED #318: LED start doing weird colors toward end of length and last 1-2 strings don't respond

wvpt, 2019-11-02:
I have 10 strings of 50 LEDS hooked together. Have power injection through entire length. When on solid color all the lights work but when doing an effect the last 1-2 strings don't work and there are some colors showing that should not be showing toward the end. I though the data wire repeats signal at every LED. Can anyone explain what is happening? Not sure if maybe there is a setting that could need to be changed?

Aircoookie, 2019-11-02:
Hi! No, you shouldn't need to change anything in settings.
I have two likely causes:
- Check the power injection. Are the wires thick enough? Maybe there is not enough voltage at the end despite of the injection. If you have a multimeter, you can measure the voltage.
- Check the connector/cable directly in front of the first malfunctioning strip. Maybe it has bad contact, which could cause something like this.

wvpt, 2019-11-05:
Does the data being sent get repeated at each led?

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

raoulteeuwen, 2019-11-05:
> Does the data being sent get repeated at each led?

I was curious, so did some Googl-ing, and found "With the WS2812B and similar LEDs, the data signal itself is regenerated by each LED chip, so that isn't generally an issue." at/in/on https://github.com/FastLED/FastLED/issues/288 .

brentbrooks70, 2019-11-08:
@raoulteeuwen said
> I was curious, so did some Googl-ing, and found "With the WS2812B and similar LEDs, the data signal itself is regenerated by each LED chip, so that isn't generally an issue." at/in/on [FastLED/FastLED#288](https://github.com/FastLED/FastLED/issues/288) .

That is exactly correct, that is why there is a _skip first LED_ at the bottom of the LED Preferences. Basically this lets you connect 1 LED in close proximity to the MCU and then run your data cable out to the actual strip. This behaves like (maybe even better than) a Logic Level Shifter

@Aircoookie said
> Check the power injection. Are the wires thick enough? Maybe there is not enough voltage at the end despite of the injection. If you have a multimeter, you can measure the voltage.

When running 5V more than a few feet you need very large diameter cables. I use regular Household AC 10 gauge wires for runs up to 15feet with an end voltage of ~4V with 30A load. Anything over 20feet and you should go to 10 gauge or larger. Of course, only use solid copper wires, aluminum loses about 40% more than copper of the same length and gauge; twisted cables of the same gauge are about 30% smaller than solid (that's the space between the strands).

Check out this [voltage drop calculator](https://www.calculator.net/voltage-drop-calculator.html?material=copper&wiresize=3.277&voltage=5&phase=dc&noofconductor=1&distance=15&distanceunit=feet&amperes=30&x=70&y=15). It has saved me numerous times, Good Luck!

closed 2020-04-05 by stale[bot]